commaai/openpilot #29867: OpenPilot got stuck in a boot loop on a C3 running master-ci

nabeelr, 2023-09-10:
### Describe the bug

I had been running master-ci for a while.

On Friday September 8, 2023 at around 16:00 GMT -4, after parking I noticed the "Connect" indicator was yellow.

To attempt to fix the issue, I rebooted my C3 via the UI.

After rebooting, the spinning comma logo seemed to stall near the end of the progress bar, then the spinner froze, and eventually my C3 rebooted. This happened 2-3 times, at which point, I unplugged my C3 from my car, waited approximately 1 minute (until the Panda's LED seemed to go out completely), plugged my C3 in again, at which point it continued boot looping in the same way.

At that point, I used flash.comma.ai to re-flash my C3 so I could use it on the way home, assuming I had just pulled a bad copy of master-ci, and since I had been looking for an excuse to try using flash.comma.ai anyways.

I assume any logs or whatnot would have been wiped as a result of re-flashing my C3, but I'd be willing to do any digging to try to recover something, if wanted.

Unfortunately, all my drives and routes that day seem to have been erased after using flash.comma.ai, and since connect was not working either, nothing seems to have been uploaded until after using flash.comma.ai.

### Provide a route where the issue occurs

unknown

### openpilot version

master-ci potentially after e8705e8f91c74ac6fc3d30c9ec9df7f4490d9553, but before 1463e81950395b712712fcd85cfb867db8343602, inclusively.

adeebshihadeh, 2023-09-29:
Gonna track this in #29211. Didn't find anything specific in your logs though

closed 2023-09-29 by adeebshihadeh

nabeelr, 2023-09-29:
Sounds good. FWIW, it hasn't happened since, but if it does again, I'll be sure to save the logs and post it in #29211