commaai/openpilot #29882: macOS: linguist

royjr, 2023-09-12:
### Describe the bug

When trying to run `linguist selfdrive/ui/translations/your_language_file.ts` as stated [here](https://github.com/commaai/openpilot/blob/master/selfdrive/ui/translations/README.md), I get `zsh: command not found: linguist`

Is mac_setup supposed to install linguist, or is it somewhere hidden in qt@5?

Installing `brew install --cask qt-creator` did not seem to help.

I found some prebuilt binaries here that worked. https://github.com/lelegard/qtlinguist-installers/releases

### OS Version

Ventura 13.5.2

### openpilot version or commit

master

fredyshox, 2023-09-12:
@royjr Does `Linguist` (capitalized) work? Is your file system case-sensitive (can check that in Disk Utility app)?

royjr, 2023-09-12:
Doesn't look like it. `APFS (Encrypted)`

```
blah@Someones-MacBook-Pro ~ % which ping
/sbin/ping
blah@Someones-MacBook-Pro ~ % which Ping
/sbin/Ping
blah@Someones-MacBook-Pro ~ % which linguist
linguist not found
blah@Someones-MacBook-Pro ~ % which Linguist
Linguist not found
```

fredyshox, 2023-09-12:
Seems like homebrew distribution of qt@5 is packaging linguist as .app bundle, which is located at `$(brew --prefix qt@5)/libexec/Linguist.app`. It can be accessed from terminal using:

```
open "$(brew --prefix qt@5)/libexec/Linguist.app" --args selfdrive/ui/translations/your_language_file.ts
```

royjr, 2023-09-13:
That works! Any way to make it an alias in mac_setup so that the instructions make sense or should the instructions just include that command for mac users?

adeebshihadeh, 2023-09-13:
We're considering re-focusing our macOS efforts on making the devcontainer experience good. Want to try that and post any feedback or issues? You can get started here: https://github.com/commaai/openpilot/tree/master/tools#dev-container

adeebshihadeh, 2023-09-20:
Recommended setup is now to use the Dev Containers for non-20.04: https://github.com/commaai/openpilot/tree/master/tools#dev-container-on-any-linux-or-macos

closed 2023-09-20 by adeebshihadeh