commaai/openpilot #29883: Run openpilot on Ubuntu 22.04 desktop PC under aarch64 architecture

L-Net-1992, 2023-09-12:
### Describe the bug

I want to run openpilot on Ubuntu 22.04 desktop PC under **aarch64** architecture. Where can I start compiling? according to https://github.com/commaai/openpilot/blob/live-magd/tools/README.md Is this an explanation?To provide a detailed description, I have an embedded development board for RK3588 that is installed with the ubuntu22.04 system. However, as it is based on the aarch64 architecture, I do not know how to start compiling or modify it based on which document or script. It can run on RK3588. As for the model, how to run on GPU or NPU will be discussed later.

### OS Version

Ubuntu 22.04

adeebshihadeh, 2023-09-12:
openpilot already compiles for aarch64, and we even test that in CI. Are you having an issue just building it?

L-Net-1992, 2023-09-12:
> openpilot already compiles for aarch64, and we even test that in CI. Are you having an issue just building it?

I have just come into contact with this and don't know much about it. Excuse me, is it according to https://github.com/commaai/openpilot/blob/master/tools/README.md Is this document operational? If not, please provide some feedback. Thank you!

adeebshihadeh, 2023-09-12:
Our issue tracker is for bugs. If you'd just like to discuss this, open a discussion in the Discussions tab instead.

closed 2023-09-12 by adeebshihadeh