commaai/openpilot #29905: Getting car unrecognized message on supported car

Quake4net, 2023-09-13:
### Describe the bug

[image]
[image]
[image]

### Which car does this affect?

VW Crafter 2017

### Provide a route where the issue occurs

No route because car was not recognized

### openpilot version

0.9.4

sshane, 2023-09-15:
@jyoung8607 want to handle this?

```
{'carParams': {'alternativeExperience': 1,
               'autoResumeSng': True,
               'carFingerprint': 'mock',
               'carFw': [{'address': 1879,
                          'brand': 'volkswagen',
                          'bus': 1,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x872Q0907572J \xf1\x890156',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x87\xf1\x89\xf1\x82'],
                          'responseAddress': 1985,
                          'subAddress': 0},
                         {'address': 1810,
                          'brand': 'volkswagen',
                          'bus': 1,
                          'ecu': 'eps',
                          'fwVersion': b'\xf1\x872N0909143D\x00\xf1\x897010\xf1\x82\x05183AZ306A2',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x87\xf1\x89\xf1\x82'],
                          'responseAddress': 1916,
                          'subAddress': 0},
                         {'address': 1813,
                          'brand': 'volkswagen',
                          'bus': 1,
                          'ecu': 'srs',
                          'fwVersion': b'\xf1\x873Q0959655AL\xf1\x890505\xf1\x82\x0e1411001413001203151311031100',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x87\xf1\x89\xf1\x82'],
                          'responseAddress': 1919,
                          'subAddress': 0},
                         {'address': 2016,
                          'brand': 'volkswagen',
                          'bus': 1,
                          'ecu': 'engine',
                          'fwVersion': b'\xf1\x8704L906056BP\xf1\x894729',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x87\xf1\x89\xf1\x82'],
                          'responseAddress': 2024,
                          'subAddress': 0},
                         {'address': 2016,
                          'brand': 'hyundai',
                          'bus': 1,
                          'ecu': 'engine',
                          'fwVersion': b'\xf1\x8704L906056BP',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x87\xf1\x81\xf1\x00'],
                          'responseAddress': 2024,
                          'subAddress': 0}],
```

Quake4net, 2023-09-15:
I did it myself (thanks to the discord community!):
https://github.com/commaai/openpilot/pull/29926

closed 2023-09-20 by sshane