commaai/openpilot #29911: Is support available for “Kia Optima Hybrid 2019”?

rockimjh, 2023-09-14:
### Describe the bug

Is support available for “Kia Optima Hybrid 2019”?
At comma 3x, the vehicle is not recognized and starts in dashcam mode.
So I tried extracting my car's fingerprint and firmware information and adding them to value.py, but it still goes into dashcam mode.
I am sharing the fingerprint and firmware information I extracted for reference, and if you have any information you need, please let me know.

### Which car does this affect?

Kia Optima Hybrid 2019

### Provide a route where the issue occurs

6a42c1197b2a8179|2023-09-14--12-12-02

### openpilot version

0.9.5

### Additional info

- Fingerprint information
68: 8, 127: 8, 304: 8, 320: 8, 339: 8, 352: 8, 356: 4, 544: 8, 546: 8, 548: 8, 549: 8, 576: 8, 593: 8, 688: 5, 832: 8, 881: 8, 882: 8, 897: 8, 902: 8, 903: 8, 909: 8, 916: 8, 1040: 8, 1056: 8, 1057: 8, 1078: 4, 1136: 6, 1151: 6, 1157: 4, 1168: 7, 1173: 8, 1180: 8, 1186: 2, 1191: 2, 1236: 2, 1265: 4, 1280: 1, 1287: 4, 1290: 8, 1291: 8, 1292: 8, 1294: 8, 1322: 8, 1342: 6, 1345: 8, 1348: 8, 1355: 8, 1363: 8, 1369: 8, 1371: 8, 1397: 8, 1407: 8, 1419: 8, 1427: 6, 1429: 8, 1430: 8, 1448: 8, 1456: 4, 1470: 8, 1476: 8, 1535: 8, 2000: 8, 2004: 8, 2005: 8, 2008: 8, 2012: 8, 2013: 8

- Vehicle firmware information
                'carFw': [{'address': 2000,
                           'brand': 'hyundai',
                           'bus': 1,
                           'ecu': 'fwdRadar',
                           'fwVersion': b'\xf1\x00JFhe SCC FHCUP 1.00 1.01 99110-A8500 ',
                           'logging': False,
                           'obdMultiplexing': True,
                           'request': [b'"\xf1\x00'],
                           'responseAddress': 2008,
                           'subAddress': 0},
                          {'address': 1988,
                           'brand': 'hyundai',
                           'bus': 1,
                           'ecu': 'fwdCamera',
                           'fwVersion': b'\xf1\x00JFH MFC AT KOR LHD 1.00 1.01 95895-A8200 180323',
                           'logging': False,
                           'obdMultiplexing': True,
                           'request': [b'"\xf1\x00'],
                           'responseAddress': 1996;
                           'subAddress': 0},
                          {'address': 2016,
                           'brand': 'hyundai',
                           'bus': 1,
                           'ecu': 'engine',
                           'fwVersion': b'\xf1\x816H6D1051\x00\x00\x00\x00\x00\x00\x00\x00',
                           'logging': False,
                           'obdMultiplexing': True,
                           'request': [b'"\xf1\x87\xf1\x81\xf1\x00'],
                           'responseAddress': 2024;
                           'subAddress': 0},
                          {'address': 2000,
                           'brand': 'hyundai',
                           'bus': 1,
                           'ecu': 'fwdRadar',
                           'fwVersion': b'\xf1\x00JFhe SCC FHCUP 1.00 1.01 99110-A8500 ',
                           'logging': False,
                           'obdMultiplexing': True,
                           'request': [b'"\xf1\x87\xf1\x81\xf1\x00'],
                           'responseAddress': 2008,
                           'subAddress': 0},
                          {'address': 1975,
                           'brand': 'hyundai',
                           'bus': 0,
                           'ecu': 'cornerRadar',
                           'fwVersion': b'\xf1\x00\xff\x00',
                           'logging': True,
                           'obdMultiplexing': True,
                           'request': [b'"\xf1\x00'],
                           'responseAddress': 1983,
                           'subAddress': 0},
                          {'address': 2000,
                           'brand': 'hyundai',
                           'bus': 0,
                           'ecu': 'fwdRadar',
                           'fwVersion': b'\xf1\x00JFhe SCC FHCUP 1.00 1.01 99110-A8500 ',
                           'logging': False,
                           'obdMultiplexing': True,
                           'request': [b'"\xf1\x00'],
                           'responseAddress': 2008,
                           'subAddress': 0},
                          {'address': 1988,
                           'brand': 'hyundai',
                           'bus': 0,
                           'ecu': 'fwdCamera',
                           'fwVersion': b'\xf1\x00JFH MFC AT KOR LHD 1.00 1.01 95895-A8200 180323',
                           'logging': False,
                           'obdMultiplexing': True,
                           'request': [b'"\xf1\x00'],
                           'responseAddress': 1996;
                           'subAddress': 0},

sshane, 2023-09-14:
Looks like you figured it out, you can open a pull request or let me know if you need help.

sshane, 2023-09-14:
The only other user of an Optima Hybrid had an issue with their SCC checksum from the radar being different so we put it behind dashcam until we understood it. However your car seems to be newer by 2 years (his is a 2017 I believe), and I didn't see the same failure in test_models. Relevant PR: https://github.com/commaai/openpilot/pull/29520

Can you get a longer route, and we can add you under a new platform?

rockimjh, 2023-09-15:
I installed a new software using the Installer URL you provided, connected it to the vehicle, and ran it, but I saw "Can Error: Check connections" occur. And I uploaded a longer route driven in this state. Please refer to this route and make sure that my car is supported.
route results: 6a42c1197b2a8179|2023-09-15--20-38-08
[image]

sshane, 2023-09-15:
Sorry about that, please try again. Make sure to do the following:

```
Can you get a longer route, and we can add you under a new platform? Make sure the route starts and ends a public place and is about 10 minutes, as we will make the route public for CI testing.
```

rockimjh, 2023-09-16:
I drove the longer route like you said, over 10 minutes, and I'm sending you the results. (with CAN Error status)
I think I probably didn't give you access to the route information I wrote in my previous post, so I've now given you access to your email address (shane@smiskol.com) so you can see it.
This is my first experience using a comma device, so I may not know much about transmitting driving data to you, so please let me know if there is any additional action required.

new results: 6a42c1197b2a8179|2023-09-16--17-29-49

rockimjh, 2023-09-16:
Thank You!!! The “CAN Error” message has disappeared. I've driven it for about 10 minutes and it seems to be working well. However, this is my first time using OpenPilot, so I'm not sure if all the functions are working properly. As you said, I am leaving a driving log, so if there is anything that needs to be corrected, please do so.

result: 6a42c1197b2a8179|2023-09-16--18-59-44

sshane, 2023-09-20:
Which harness did you use?

rockimjh, 2023-09-21:
Hyundai H

closed 2023-09-22 by sshane

amontanye, 2023-11-27:
I have a 2017 Kia Optima PHEV. Will Comma 3X be compatible with my vehicle?
*The website says it needs ACC which it has but the discord has stated otherwise.*

I feel like 2017/2019 Hybrid is the closest I've seen to my vehicle.