commaai/openpilot #29912: sentry mode

adeebshihadeh, 2023-09-14:
v1: records a 1m route, big alert on screen, and maybe sends a text

jakethesnake420, 2023-10-13:
My car has been broken into twice and they didn't take the C3. I feel like if it wakes up and draws attention people will be more likely to take it. I think there should be a silent sentry toggle

morrislee, 2023-10-20:
> My car has been broken into twice and they didn't take the C3. I feel like if it wakes up and draws attention people will be more likely to take it. I think there should be a silent sentry toggle

maybe dont wake up the device display, but just do a cloud push alert to the registered account via email to see a live stream and then you can action how you wish?

royjr, 2023-10-22:
Live uploading would work better in cases of theft than waiting to upload a 1-minute video. If the device is unplugged or stolen before it sends the video, it's no good. A thief would probably be gone before the minute is recorded.