commaai/openpilot #29921: Hyundai CAN (EV): TCS13->DriverBraking doesn't describe brake pedal depressed status

sshane, 2023-09-15:
It seems to describe whether the user has depressed far enough to actuate friction brakes. Before then it's using regen braking which is a different signal (that also has gas pressed!).

[image]

sshane, 2023-09-15:
@briantran33 can you get a route in dashcam mode with stock SCC activated and lightly and slowly press the brakes until it disengages? Try to test both when the stock system is accelerating an braking

sshane, 2023-09-15:
On our EV6 it looks like the TCS's DriverBraking is actually DriverOverride (without the gas pressed status), as the brake bit from `BRAKE` looks similar to the CAN's version (doesn't capture low sens inputs). So only an issue on CAN EVs.

[image]

briantran33, 2023-09-15:
> @briantran33 can you get a route in dashcam mode with stock SCC activated and lightly and slowly press the brakes until it disengages? Try to test both when the stock system is accelerating and braking

Sure, I will do that after I get back from the vacation.

briantran33, 2023-09-23:
d3a039910163c67a|2023-09-22--22-09-24--0 - This should have both the scenarios. @sshane

sshane, 2023-09-26:
It indeed looks like SCC uses DriverOverride, not DriverBraking!

[image]

sshane, 2023-11-09:
Same with our Sonata (stock route):

ee11091550c9a627|2023-11-08--21-49-12

[image]

closed 2023-11-10 by sshane