commaai/openpilot #29948: ev6 w/o HDAII doesn't display FCA/AEB warning when in OP long

bdimcheff, 2023-09-16:
### Describe the bug

It's my understanding (based on the warning message about OP long being alpha) that on an ev6 with HDA I, enabling OP longitudinal control will disable stock AEB and FCA. The indicator on the dashboard for my ev6 definitely does not have any warning lights illuminated, however. I don't know if this means that stock AEB/FCA is still enabled, or if it is disabled but the warning lights aren't properly illuminating.

### Which car does this affect?

Kia EV6 2023 Wind AWD (HDA I)

### Provide a route where the issue occurs

fab807792f63fea6|2023-09-15--18-57-23--0

### openpilot version

master-ci

sshane, 2023-09-18:
It indeed should be showing for HDA1 cars, I think we forgot about this when merging https://github.com/commaai/openpilot/pull/26345.

Can you get two routes: one with AEB disabled in the settings so it shows [this icon](https://www.reddit.com/media?url=https%3A%2F%2Fpreview.redd.it%2Ftjuu2ls56tua1.jpg%3Fwidth%3D960%26crop%3Dsmart%26auto%3Dwebp%26s%3D9dd35e3dd37a562122330f50e348c7232ed35844), and one with AEB enabled as normal?

bdimcheff, 2023-09-20:
@sshane yeah sure. I assume this means to disable it via the car's control settings and just drive for a minute. Do I need to actually activate openpilot, or just route logging while I'm driving is enough?

sshane, 2023-09-20:
Just 1 minute parked should be enough.

bdimcheff, 2023-09-22:
aeb enabled/default settings on `fab807792f63fea6|2023-09-22--19-37-45--0`.

aeb disabled for the latter part of `fab807792f63fea6|2023-09-22--19-41-46--0`. It seems like you can't disable it permanently, as it toggles back on when the car is power cycled.

sshane, 2023-09-23:
Can you try this? https://github.com/commaai/openpilot/pull/30021