commaai/openpilot #29964: proclogd: random test failures

adeebshihadeh, 2023-09-19:
Seems to have started when we moved to pytest-cpp.

An example - https://github.com/commaai/openpilot/actions/runs/6229262650/job/16907446278
```
=================================== FAILURES ===================================
_______________________________ Parser::procStat _______________________________
[gw0] linux -- Python 3.11.4 /root/.pyenv/versions/3.11.4/bin/python3.11
Expected: 
            parsed_cnt == pids.size()
          
Actual: 
            5 == 6
          
system/proclogd/tests/test_proclog.cc:44: C++ failure

---------- coverage: platform linux, python 3.11.4-final-0 -----------
Coverage XML written to file coverage.xml
```

Got some different ones while running `time while true; do pytest . || break; done`

adeebshihadeh, 2023-09-19:
@deanlee want to take a look?

deanlee, 2023-09-19:
found two related issue:

1. pid may become invalid before read stat from `/proc/pid/stat`.
2. pytest changes the command line to do different tests. for example, the same pid may have different command line at different time:

first snapshot:
```
  "[pytest-xdist running] selfdrive/car/tests/test_lateral_limits.py::
  TestLateralLimits_49_HONDA_FREED_2020::test_jerk_limits"
```
second snapshot.
```
  "[pytest-xdist running] selfdrive/car/tests/test_lateral_limits.py::
  TestLateralLimits_49_HONDA_FREED_2020::test_jerk_limits"
```

I'm trying to fix these two issues.

closed 2023-09-21 by adeebshihadeh