commaai/openpilot #29986: Unrecognized car - Kia Sportage 2023

kiwicolt, 2023-09-20:
### Describe the bug

I've just received my comma 3x and am getting unrecognized car error.

Kia Sportage 2023.

Dongle ID c1b14f8db6a238cf

### Which car does this affect?

Kia Sportage 2023

### Provide a route where the issue occurs

n/a

### openpilot version

0.9.4

kiwicolt, 2023-09-20:
route =c1b14f8db6a238cf|2023-09-20--13-58-26--0

kiwicolt, 2023-09-20:
{'carParams': {'alternativeExperience': 1,
               'autoResumeSng': True,
               'carFingerprint': 'mock',
               'carFw': [{'address': 1975,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'cornerRadar',
                          'fwVersion': b'\xf1\x00NQ5 BCW RR 1.00 , 1.01 (uR\x89"\x00\x82\x00\x08',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1983,
                          'subAddress': 0},
                         {'address': 1988,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00NQ5 FR_CMR AT AUS RHD 1.00 1.00 99211-P1040 663',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1996,
                          'subAddress': 0},
                         {'address': 1971,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'hvac',
                          'fwVersion': b'\xf1\x00NQ5   97255-P1500UNIT-HEATER CONTROL 1.01 NQ5 DATC 1.3 1.00   ',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1979,
                          'subAddress': 0},
                         {'address': 2000,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00NQ5__               1.00 1.03 99110-P1000         ',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 2008,
                          'subAddress': 0}],

sshane, 2023-09-20:
Fixed in https://github.com/commaai/openpilot/pull/29990