commaai/openpilot #29989: Hyundai CAN FD: use less of the part number for fuzzy FP

sshane, 2023-09-20:
We're getting PRs like [this](https://github.com/commaai/openpilot/pull/29962/files) and [this](https://github.com/commaai/openpilot/pull/29988) where the part number's suffix increments indicating some sort of part revision, but the major first two characters remain the same. Explore reducing the part number for fingerprinting to just the first characters. It still catches the Tucson with the different tuning, as well as https://github.com/commaai/openpilot/pull/29949 which we don't want to merge for the same reason.

https://github.com/commaai/openpilot/issues/29572

sshane, 2023-09-23:
https://github.com/commaai/openpilot/pull/30020 would also fingerprint with this.

sshane, 2023-10-10:
Something like this: https://github.com/commaai/openpilot/commit/47207e246404004eb6c34a488efcfd8eb2738922

sshane, 2023-10-10:
Though we do have to be careful. There's an example of the same two characters in the part number for a new 2024 Elantra fwdRadar, however they switched from having it send SCC to just being a radar. The only thing that would prevent it from matching with this is the date code.

dongle: 784a39c0a9dd3493

```python
# 2021
b'\xf1\x00CN7_ SCC FNCUP      1.00 1.01 99110-AA000         ',
# 2024
b'\xf1\x00CN7_ RDR -----      1.00 1.01 99110-AA500         ',
```

gregjhogan, 2023-10-10:
Fuzzy fingerprinting on things we understand (platform codes, dates, software version numbers) seems fairly safe.  I am not sure how we make any safe assumptions about the future if extend to components we don't understand what the values represent.

sshane, 2024-03-30:
We don't understand the part numbers enough yet for this. Can revisit in the future

closed 2024-03-30 by sshane