commaai/openpilot #29998: stop and go working intermittently

thealexbaron, 2023-09-21:
### Describe the bug

Right around 15:11:36 – 7 in the shared route, the Jeep in front of me starts to move. I had to press the accelerator pedal to get the car to move even though open pilot was engaged.

### Which car does this affect?

Ioniq 6 2023

### Provide a route where the issue occurs

a3156f49719f7455|2023-09-18--15-04-23--0

### openpilot version

0.9.5

### Additional info

Stop and go appears to be effective about half of the time.

sshane, 2023-09-21:
Those logs are gone already, can you send a more recent example so I can request upload on full logs?

sshane, 2023-09-26:
@thealexbaron any new examples? I couldn't find any recent ones

adeebshihadeh, 2024-01-05:
Closing due to inactivity

closed 2024-01-05 by adeebshihadeh