commaai/openpilot #29999: FCW warning alerts when long control enabled

hletrd, 2023-09-21:
### Describe the bug

I've just wrote a new car port for my Kia K5 Hybrid 2022. It worked well for a while with openpilot(```master``` branch) and sunnypilot(```dev-c3```), happily testing E2E long features.

But after a few hours, the FCW warning indicator started to show up on my dashboard. The warning indicator does not appear when openpilot long control is disabled, but consistently appear when long control is enabled. No error log on console (when I attached to the tmux.) Additionally, the control buttons on my steering wheel does not work properly. (Nothing but only +/- button are working.) The speed limit is always being set to 65mph/105kph every time I engage openpilot. I tried to replug all connector, reflash my comma 3X but nothing changes.

### Which car does this affect?

Kia K5 Hybrid 2022

### Provide a route where the issue occurs

not uploaded yet but will check soon

### openpilot version

0857f4b

speedking456, 2023-09-21:
Based on what you've described, this all sounds like intended behaviour. When OP long is toggled on, the FCW warning will be present (as it disables the stock FCW), cruise will always be active, pressing the cruise button will not toggle anything, the follow distance button will not adjust anything, and only +/- and set/res on the same rocker switch will work. The speed limit being set higher is likely due to experimental mode being enabled; it uses the driving model to decide what speed to travel and sets the set speed to be closer to the max highway speed. With experimental disabled, the set/res should work like stock, and with OP long disabled it will use the stock SCC system and only take over lateral control when cruise is set.

Hope this clears that up, if not, please upload a drive so someone can take a look.

sshane, 2023-09-21:
Yes, we purposefully show this icon when using alpha longitudinal control (regardless of experimental mode or not):

[image]

If you see any other warnings, that is not normal. Can you send a photo of your dash and send a route where your steering wheel buttons don't work? What do you mean by that?

sshane, 2023-09-22:
@hletrd I think you're describing that the MAIN button and/or pause/resume buttons don't work to enable openpilot if using alpha longitudinal? If so that is also expected, as openpilot expects resume or set.

Please send a photo of the "FCW warning" that only shows sometimes to confirm it's the normal one and not a dash fault.

hletrd, 2023-09-22:
Got it. There is no other unexpected behavior, or any warning indicator showing up.

sshane, 2023-09-22:
Thanks for confirming, closing.

closed 2023-09-22 by sshane