commaai/openpilot #30011: Toyota Highlander XLE 2021 not recognized

Vorne12, 2023-09-22:
### Describe the bug

Comma 3 was working perfectly for over a year until it suddenly stopped recognizing the car a few weeks ago and it only worked in Dashcam mode from then on. It says it's "unable to identify your car".

I checked all connections multiple times, reset and re-uploaded standard software repeatedly with no success. Based on other suggestions, i uploaded master-ci (https://installer.comma.ai/commaai/master-ci) which also didn't help.

Any suggestions what the next steps are to fix it?

Mahalo....

[image]

### Which car does this affect?

Toyota Highlander 2021

### Provide a route where the issue occurs

6c4eb3920653c33f / 7dd1c82d

### openpilot version

0.9.6/master-ci/b7ca539/Sep 22

sshane, 2023-09-22:
I see some errors from the panda after it stops fingerprinting due to no FW responses. Can you try a different/new OBD-C (USB-C) cable from the comma three to the harness box?

sshane, 2023-09-22:
@briskspirit I see `canCoreResetCnt` continually rising along with errors. False positive trigger or expected from a possibly bad cable here? 10Hz pandaStates route: 6c4eb3920653c33f|2023-09-22--18-39-50

Vorne12, 2023-09-22:
I see. Thank you. Ordered a new cable at Comma shop and hope it will work. Will let you know....

RPalmr, 2023-09-24:
Did it work?

Vorne12, 2023-09-24:
Still waiting for the cable to be shipped….

 

But thanks for following up. Will let you know….

 

Rolf

Vorne12, 2023-10-01:
Got the new cable and it looks like the problem his fixed now.

Thank you for your help, everyone   : )

closed 2023-10-01 by adeebshihadeh