commaai/openpilot #30070: Acceletometer, Gyroscope timing spikes

nuwandavek, 2023-09-27:
### Describe the bug

There are random spikes in diff(logMonoTime) for the accelerometer and gyroscope. This can cause them to be fail the `sm.allAliveAndValid` or `sm.AllAlive` checks sometimes. 

[image]

### Provide a route where the issue occurs

4c00c8dba800b92b|2023-09-25--22-15-33/201

### openpilot version

0.9.5

devtekve, 2023-10-19:
oh yeah, I've experienced this on a C3X device, not sure if C3 are the same

adeebshihadeh, 2024-02-08:
@nuwandavek I think I know what this is; can you do a quick check to see if there's a difference between C3 and 3X?

adeebshihadeh, 2024-09-06:
Bounty is for fixing the timings of `gyroscope` and `accelerometer`. 

Requirements for the bounty
* over a few hours of onroad time, max diff of both services is <20ms (2x expected)
* add a timings check for both services to [test_onroad](https://github.com/commaai/openpilot/blob/51ebf44f486ea9c26e9328c8d43fbd01ea68f53c/selfdrive/test/test_onroad.py#L80)