commaai/openpilot #30078: 2023 Ford Maverick Not Recognized

seanaguinaga, 2023-09-27:
### Describe the bug

Car Not Recognized

Looks like all the values for firmware are the same as what is already in values.py

### Which car does this affect?

Ford Maverick 2023

### Provide a route where the issue occurs

66833387c2bbbca0|2023-09-27--21-13-05

### openpilot version

master-ci

### Additional info

`{'carParams': {'alternativeExperience': 1,
               'autoResumeSng': True,
               'carFingerprint': 'mock',
               'carFw': [{'address': 2016,
                          'brand': 'ford',
                          'bus': 1,
                          'ecu': 'engine',
                          'fwVersion': b'PZ6A-14C204-JE\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 2024,
                          'subAddress': 0},
                         {'address': 1840,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'eps',
                          'fwVersion': b'NZ6C-14D003-AL\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1848,
                          'subAddress': 0},
                         {'address': 1888,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'abs',
                          'fwVersion': b'PZ6C-2D053-ED\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1896,
                          'subAddress': 0},
                         {'address': 1798,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'NZ6T-14F397-AC\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1806,
                          'subAddress': 0},
                         {'address': 1842,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'shiftByWire',
                          'fwVersion': b'NZ6P-14G395-AD\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': True,
                          'obdMultiplexing': True,
[... 273 lines cut ...]
               'safetyModelPassiveDEPRECATED': 'silent',
               'safetyParamDEPRECATED': 0,
               'startAccel': 0.0,
               'startingAccelRateDEPRECATED': 0.0,
               'startingState': False,
               'steerActuatorDelay': 0.0,
               'steerControlType': 'torque',
               'steerLimitAlert': False,
               'steerLimitTimer': 1.0,
               'steerRateCostDEPRECATED': 0.0,
               'steerRatio': 13.0,
               'steerRatioRear': 0.0,
               'stopAccel': -2.0,
               'stoppingControl': True,
               'stoppingDecelRate': 0.800000011920929,
               'tireStiffnessFactor': 1.0,
               'tireStiffnessFront': 201087.203125,
               'tireStiffnessRear': 317877.90625,
               'transmissionType': 'unknown',
               'vEgoStarting': 0.5,
               'vEgoStopping': 0.5,
               'wheelSpeedFactor': 1.0,
               'wheelbase': 2.700000047683716},
 'logMonoTime': 1021914306894,
 'valid': True}`

sshane, 2023-09-27:
We're missing your engine FW which caused it to not fingerprint. You can open a PR to add it yourself: `b'PZ6A-14C204-JE\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',` to this area:

https://github.com/commaai/openpilot/blob/dbd8e24a17f99ebc20bf674fbafabdfb0d1f5329/selfdrive/car/ford/values.py#L264

We plan on improve fuzzy fingerprinting for Ford in the near future, so cases like these don't happen.

seanaguinaga, 2023-09-28:
Good eye - this is why you have this job and I don't!

I will add and PR

Thank you

seanaguinaga, 2023-09-28:
Tried to test install but anything based on master just hangs at 92% indefinitely

seanaguinaga, 2023-09-28:
Is it normal to do that? Should we fork based on master-ci?

seanaguinaga, 2023-09-28:
It successfully installed after rebooting and starting over

closed 2023-09-28 by sshane