commaai/openpilot #30081: 2018 Honda CRV not recognized

achanin00, 2023-09-28:
### Describe the bug

New install into a 2018 Honda CRV with touring option.  The car natively has radar, line detection camera, and ACC.  OpenPilot is not recognizing the vehicle.  Please see the pictures below.

[image]
[image]
[image]
[image]

### Which car does this affect?

Honda CRV 2018

### Provide a route where the issue occurs

N/A

### openpilot version

latest downloaded from openpilot.comma.ai

### Additional info

N/A

sshane, 2023-09-28:
Have you plugged in the comma power? It's required at this time to fingerprint.

achanin00, 2023-09-28:
I am not sure what you mean by commapower.

I followed the installation video exactly. Device clearly has power since
the screen is working

sshane, 2023-09-28:
This should have come with the device. It plugs into the OBD port of the car and connects to the harness box with the RJ45 cable https://www.comma.ai/shop/comma-power

Is it plugged in?

achanin00, 2023-09-28:
Yes, of course it is plugged in. I assumed the comma power is coming from
that plug.

sshane, 2023-09-28:
I see errors on canState1 when it's multiplexed to the OBD port, as well as transmit errors, so somewhere it's not able to communicate with the car's ECUs. Can you snap a photo of it installed?

I would try:

- Swap out OBD C (USB C) cable and/or ethernet cable to harness box (try OBD C first)
- Switch to master which has better logging
- If neither work, open a support ticket at https://comma.ai/support and we can migrate the issue over to there.

achanin00, 2023-09-29:
I am traveling today.  Will take a look at the wiring when I get home
tomorrow.

The comma device is getting power.   Doesn't that imply that the USB-C
wiring is working correctly?

achanin00, 2023-09-30:
Shane,

Thank you for your help.  The issue was with iffy connection of the
ethernet cable.  Openpilot now properly recognizes my car.

Thanks again,
Alex

closed 2023-09-30 by adeebshihadeh