commaai/openpilot #30102: Process Unavailable Delter

alan-polk, 2023-09-29:
### Describe the bug

Cruise control would not engaged, error would pop up "Process Unavailable deter"

### Provide a route where the issue occurs

24574459dd7fb3e0|2023-09-29--00-06-58--0

### openpilot version

https://github.com/rwxr-xr-x/openpilot/commit/ac97ef3d9162020558f5b3597823e0315fd4c715

### Additional info

crashed on the previous route (24574459dd7fb3e0|2023-09-28--23-39-47--0) and still would don't work after a power cycle of the unit.

alan-polk, 2023-09-29:
[CrashLogs.zip](https://github.com/commaai/openpilot/files/12768525/CrashLogs.zip)

adeebshihadeh, 2023-09-29:
Looks like you wrote some stuff to that directory it's not designed to clean up (`/data/media/0/realdata/2023-09-23.tar.gz`). We'll get that fixed up so it's more robust, but it would be helpful to note that upfront.

alan-polk, 2023-09-29:
I had forgotten to clean that file up, but yes it would be helpful if it would delete the most recent drive data rather than flat out crashing.  I have cleaned that up now.  Thanks

adeebshihadeh, 2023-09-29:
@jnewb1 can you write a good test for this then fix it up? We should test lots of combos of files/symlink/directories etc.

adeebshihadeh, 2023-12-12:
Bounty is for:
* a good unit test that creates lots of files and directory with different structures
* making deleter handle any failures of the test

lspahija, 2024-02-04:
Hey, I've opened a [PR](https://github.com/commaai/openpilot/pull/31293) with a fix for this.

lspahija, 2024-02-07:
@jnewb1 @adeebshihadeh 
mind having a look at the PR when you have a minute?

wgenh, 2024-03-16:
Can someone check my [attempt](https://github.com/commaai/openpilot/pull/31892)?

kia8112, 2024-07-16:
@adeebshihadeh
can you review my PR please ?

raf-fonseca, 2025-04-18:
@alan-polk or @sshane is anybody working on this? I would like to take it up

alan-polk, 2025-04-18:
> [@alan-polk](https://github.com/alan-polk) or [@sshane](https://github.com/sshane) is anybody working on this? I would like to take it up

I do not think so.

passionworkeer, 2026-03-11:
可能需要在deleter进程中添加文件来源检查。建议在删除前检查文件的创建者标识（如检查文件元数据或特定标记），或者在进程启动时记录其创建的文件列表，然后在清理时只删除列表中的文件。这样可以避免误删其他进程创建的文件。

passionworkeer, 2026-03-11:
Consider adding a file source check in the deleter process. You could track files created by the process at startup and only delete files from that list during cleanup. This would prevent accidentally removing files created by other processes.

closed 2026-04-22 by adeebshihadeh