commaai/openpilot #30121: Openpilot performs very poorly on Carla

buaali, 2023-10-01:
### Describe the bug

 Openpilot performs very poorly on Carla. 
I successfully ran openpilot on carla, but I found that the predicted trajectory kept shaking, and the vehicle did not even follow the trajectory, causing the vehicle to hit the edge at the turn and not be able to complete the turn smoothly. 
My guess is that the turning angle is too small. I would like to know if anyone has tested which map and generation point openpilot performs better. I don’t think the model performance can be so bad.

It's worth noting that I already set dual_camera=True

### OS Version

ubuntu 20.04

### openpilot version or commit

fb89a2a6dd123077ebf6ecb250ec6848c4bc1e7e

buaali, 2023-10-01:
I'm running in official docker, and using carla 0.9.14, and I find that the trajectory predictions are mostly accurate, but in carla the vehicle doesn't seem to react quickly and change direction

jnewb1, 2023-10-01:
> I'm running in official docker, and using carla 0.9.14, and I find that the trajectory predictions are mostly accurate, but in carla the vehicle doesn't seem to react quickly and change direction

We are switching the main simulator experience to be metadrive, care to try that on the latest master?

adeebshihadeh, 2023-10-03:
Closed. MetaDrive is pretty good and we just made it the default.

closed 2023-10-03 by adeebshihadeh