commaai/openpilot #30130: 2020 Hyundai Kona Limited

commakona, 2023-10-02:
My 2020 Hyundai Kona Limited Edition is displaying "Car Unrecognized" and I need assistance. I'm using Comma 3X.

Thanks,

Dennis

sshane, 2023-10-02:
d0c8a13d4a1b1125

commakona, 2023-10-02:
Yes, that is my device number.

commakona, 2023-10-02:
Thank you for taking the time to respond. Could you please assist me in getting my COMMA 3X to work with my 2020 Hyundai Kona Limited Edition?
Thanks,

Dennis

sshane, 2023-10-03:
You can open a PR to merge your FW versions here by following this: https://github.com/commaai/openpilot/wiki/Fingerprinting

Or I will get to it by the end of the week.

commakona, 2023-10-03:
Shane,

Thank you for replying to my email.

Is there any way you could call me and walk me through the process? I can
pay you, and I hope it is not too expensive. I simply cannot follow these
steps.

Thanks in advance.

Dennis

jnewb1, 2023-10-03:
try this: installer.comma.ai/commaai/kona-fingerprint

commakona, 2023-10-04:
Thank you for your response.

This is extremely difficult to follow. I'm hoping someone will call me and
walk me through the process step by step. I am willing to pay, but
hopefully not a lot of money.

Thanks,

Dennis

sshane, 2023-10-05:
@commakona It appears that your car does not have Smart Cruise Control (SCC), which is required for openpilot operation. Noticed due to the fact that no radar responded to our queries, and there's no SCC messages on CAN in your qlogs. If your car *does* have SCC, please respond back and we will look into it.

SCC is only available on the Ultimate trim as per the brochure: 
[image]

Sorry about the inconvenience, we are trying to make our car compatibility more clear, including showing a photo on the website that asks you to check your steering wheel for an SCC-specific button before ordering.

You can open a refund request if you so like at https://comma.ai/returns

sshane, 2023-10-05:
BTW this is the icon you're looking for:

[image]

closed 2023-10-05 by sshane