commaai/openpilot #30164: Comma 3. Car Unrecognized msg. Toyota Camry Hybrid 2021 (Thailand)

dwerty3, 2023-10-04:
### Describe the bug

Hello.
I bought new Comma 3, install it but it showed messages after start:
1. Openpilot was unable to identify your car. Your car is either unsupported or ECUs are not recognized. Please submit a pull request to add the firmware version to the proper vehicle. Need support? Join discourd.comma.ai.
2. Unsupported car.
3. Car Unrecognized.

Also driver camera preview button didn't work and showed "Starting camera" only and nothing more.

I read [fingerprint manual](https://github.com/commaai/openpilot/wiki/Fingerprinting) then add my car fingerprint to [values.py](https://github.com/dwerty3/openpilot/blob/master/selfdrive/car/toyota/values.py) and reinstall from https://installer.comma.ai/dwerty3/master

**Fingerprint from "carParams" log:**

               'carFw': [{'address': 1968,
                          'brand': 'toyota',
                          'bus': 0,
                          'ecu': 'abs',
                          'fwVersion': b'F152633214\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>', b'\x1a\x88\x01'],
                          'responseAddress': 1976,
                          'subAddress': 0},
                         {'address': 1937,
                          'brand': 'toyota',
                          'bus': 0,
                          'ecu': 'dsu',
                          'fwVersion': b'8821F0609100    ',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>', b'\x1a\x88\x01'],
                          'responseAddress': 1945,
                          'subAddress': 0},
                         {'address': 1920,
                          'brand': 'toyota',
                          'bus': 0,
                          'ecu': 'srs',
                          'fwVersion': b'8917F06420\x00\x00\x00\x00\x00\x00',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'>', b'\x1a\x88\x01'],
                          'responseAddress': 1928,
                          'subAddress': 0},
                         {'address': 1924,
                          'brand': 'toyota',
                          'bus': 0,
                          'ecu': 'srs',
                          'fwVersion': b'8917F06420\x00\x00\x00\x00\x00\x00',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'>', b'\x1a\x88\x01'],
                          'responseAddress': 1932,
                          'subAddress': 0},
                         {'address': 1953,
                          'brand': 'toyota',
                          'bus': 0,
                          'ecu': 'eps',
                          'fwVersion': b'8965B33621\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>', b'\x1a\x88\x01'],
                          'responseAddress': 1961,
                          'subAddress': 0},
                         {'address': 1988,
                          'brand': 'toyota',
                          'bus': 0,
                          'ecu': 'hvac',
                          'fwVersion': b'8865F3301101    ',
                          'logging': True,
[... 118 lines cut ...]
      b'\x03838000XF3128\x00\x00\x00\x00838000XF3222\x00\x00\x00\x00838000XF3322\x00\x00\x00\x00',
    ],
    (Ecu.epb, 0x750, 44): [
      b'895DF33010\x00\x00\x00\x00\x00\x00',
    ],    
  # END. notice that the firmware from car params is added here.
```

All of this doesn't help.

### Which car does this affect?

Toyota Camry Hybrid 2021 (Thailand)

### Provide a route where the issue occurs

2ed432126627c66a|2023-10-04--14-14-04

### openpilot version

0.9.5 master

### Additional info

I tried many times to reinstall and change parameters again but can't fix it. Spent already 18 months. Hope you can help me to fix it.

jnewb1, 2023-10-04:
try this
installer.comma.ai/commaai/toyota-camry-hybrid-fingerprint

sshane, 2023-10-04:
Your local model year might be 2021, but to openpilot it's a TSS-P Camry Hybrid so you had it in the wrong platform. Can you open a PR to add it to CAMRYH platform? If not just let me know. FYI https://github.com/commaai/openpilot/pull/28641 catches this

dwerty3, 2023-10-05:
> Can you open a PR to add it to CAMRYH platform? If not just let me know. FYI 

- Thank you for answer. Sorry, but I don't know how to do this. Could you explain?

sshane, 2023-10-05:
I added your missing engine FW to master. Please try `master-ci` or `nightly` until the next release!

https://installer.comma.ai/commaai/nightly

closed 2023-10-05 by sshane

dwerty3, 2023-10-12:
> I added your missing engine FW to master in  https://github.com/commaai/openpilot/pull/30177. Please try `master-ci` or `nightly` until the next release!

Hello, @sshane.

I updated it to openpilot nightly (12 Oct) but comma still show Camera Malfunction.

`
[image]

[image]

[image]

`

sshane, 2023-10-12:
You should contact support at https://comma.ai/support