commaai/openpilot #30179: scons build fail with ubuntu 20.04. conda virtual enviroment python 3.11.8

VietPT3502, 2023-10-05:
### Describe the bug

this is my log:
tools/ubuntu_setup.sh
.............................................
----   OPENPILOT SETUP DONE   ----
Open a new shell or configure your active shell env by running:
source ~/.bashrc

poetry shell:
Virtual environment already activated: /home/vietpt/miniconda3/envs/openpilot
scons -u -j$(nproc):
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
clang++ -o cereal/messaging/msgq_tests.o -c -std=c++1z -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DSWAGLOG="\"common/swaglog.h\"" -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 cereal/messaging/msgq_tests.cc
clang++ -o cereal/messaging/test_runner.o -c -std=c++1z -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DSWAGLOG="\"common/swaglog.h\"" -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 cereal/messaging/test_runner.cc
clang++ -o cereal/visionipc/test_runner.o -c -std=c++1z -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DSWAGLOG="\"common/swaglog.h\"" -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 cereal/visionipc/test_runner.cc
clang++ -o cereal/visionipc/visionipc_tests.o -c -std=c++1z -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DSWAGLOG="\"common/swaglog.h\"" -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 cereal/visionipc/visionipc_tests.cc
In file included from cereal/messaging/msgq_tests.cc:1:
third_party/catch2/include/catch2/catch.hpp:1:In file included from 1cereal/messaging/test_runner.cc:: 2error:
: third_party/catch2/include/catch2/catch.hppunknown type name 'version':
1:1: error: unknown type name 'version'
version https://git-lfs.github.com/spec/v1
^
version https://git-lfs.github.com/spec/v1
^
third_party/catch2/include/catch2/catch.hppthird_party/catch2/include/catch2/catch.hpp::11::1414::  errorerror: : expected ';' after top level declaratorexpected ';' after top level declarator

version https://git-lfs.github.com/spec/v1version https://git-lfs.github.com/spec/v1

             ^             ^

             ;             ;

third_party/catch2/include/catch2/catch.hpp:2:1third_party/catch2/include/catch2/catch.hpp:: 2error:: 1unknown type name 'oid'; did you mean 'void'?:
 error: unknown type name 'oid'; did you mean 'void'?
oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fc
^~~
void
oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fc
^~~
void
third_party/catch2/include/catch2/catch.hpp:2:5: error: variable has incomplete type 'void'
third_party/catch2/include/catch2/catch.hpp:2:5: error: variable has incomplete type 'void'
oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fc
    ^
oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fc
    ^
third_party/catch2/include/catch2/catch.hpp:2:11: error: expected ';' after top level declarator
third_party/catch2/include/catch2/catch.hpp:2:oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fc11
:          ^ 
error          ;: 
expected ';' after top level declarator
third_party/catch2/include/catch2/catch.hpp:2:12: error: expected unqualified-id
oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fc
          ^
          ;
oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fcthird_party/catch2/include/catch2/catch.hpp
:           ^2
:12: error: expected unqualified-id
oid sha256:27da57c7a06d09be8dd81fab7246b79e7892b6ae7e4e49ba8631f1d5a955e3fc
           ^
6 errors generated.
In file included from cereal/visionipc/test_runner.cc:2:
third_party/catch2/include/catch2/catch.hpp:1:1: error: unknown type name 'version'
version https://git-lfs.github.com/spec/v1
^
third_party/catch2/include/catch2/catch.hpp:1:14: error: expected ';' after top level declarator
version https://git-lfs.github.com/spec/v1
[... 211 lines cut ...]
    REQUIRE(ctx["dongle_id"].string_value() == dongle_id);
    ^
common/tests/test_swaglog.cc:54:5: error: use of undeclared identifier 'REQUIRE'
    REQUIRE(ctx["dirty"].bool_value() == true);
    ^
common/tests/test_swaglog.cc:56:5: error: use of undeclared identifier 'REQUIRE'
    REQUIRE(ctx["version"].string_value() == COMMA_VERSION);
    ^
common/tests/test_swaglog.cc:59:5: error: use of undeclared identifier 'REQUIRE'
    REQUIRE(ctx["device"].string_value() == device);
    ^
common/tests/test_swaglog.cc:62:5: error: use of undeclared identifier 'REQUIRE'
    REQUIRE((thread_id >= 0 && thread_id < thread_cnt));
    ^
common/tests/test_swaglog.cc:67:21: error: invalid operands to binary expression ('const char [9]' and 'int')
    INFO("thread :" << i);
         ~~~~~~~~~~ ^  ~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
scons: *** [common/tests/test_swaglog.o] Error 1
scons: building terminated because of errors.

### OS Version

Ubuntu 20.04

jnewb1, 2023-10-05:
we pushed several large files to lfs, try doing a git lfs pull

closed 2023-10-05 by adeebshihadeh

VietPT3502, 2023-10-06:
but i was full clone with
git clone --recurse-submodules https://github.com/commaai/openpilot.git