commaai/openpilot #30201: GM: Cruise Fault on start up

Verylukyguy, 2023-10-06:
### Describe the bug

Every time the vehicle is started, the Comma device show a "Cruise Fault: Restart the car to engage" error that resolves on its own within a few seconds.

### Provide a route where the issue occurs

df51b2195de66a9b|2023-10-06--13-36-10--0

### openpilot version

Comma Nightly @ f40b1da

sshane, 2023-10-10:
Is this recent or has it being doing this as long as you can remember?

sshane, 2023-10-10:
Yep, it cleared 1 second after we started sending our messages. I'll keep this open to check if there's any initialization signals briefly

sshane, 2023-10-21:
This fault is expected for this hardware setup, we can look into generically detecting and hiding these alerts without false positives in the future if more cars are like this. Closing for now.

closed 2023-10-21 by sshane

Guardiola31337, 2023-11-10:
I want to chime in here because I'm running into `Cruise Fault: Restart the car to engage` error but in my case it doesn't resolve and the message doesn't go away (wait a long time and even drive and it keeps showing). This on a brand new Chevy Bolt EV 2023 2LT. Worth mentioning that initially I run into `Dashcam Mode Car Unrecognized which I was able to remove after adding the fingerprints as described in https://github.com/commaai/openpilot/wiki/Fingerprinting#fingerprinting-10 i.e. replaced https://github.com/commaai/openpilot/blob/72cc2e34cbd3ad25557c4b226c12790cb97457cd/selfdrive/car/gm/values.py#L209 by the fingerprint that I got from `PYTHONPATH=/data/openpilot PREPAREONLY=1 /data/openpilot/selfdrive/debug/get_fingerprint.py`

`599ef8fe5b42e9fc|2023-11-09--23-43-30`

I tried `release3` and `master-ci` and still alert doesn't disappear. Any ideas @sshane @twilsonco?

cc @nworb-cire

sshane, 2023-11-10:
This isn't related to this issue. Does your car have the ACC package? Can you send a photo of your steering wheel?

Guardiola31337, 2023-11-10:
> Does your car already brake for lead cars when using cruise control?

Will test, as mentioned we got the car a couple of days ago and couldn't test-drive all the features yet. In any case, what are you thinking? Do you think it's an issue with the car system?

Verylukyguy, 2023-11-10:
According to the options codes for this 2023 Chevy Bolt EV LT2, it does not have ACC.

The vehicle needs to be an LT2 trim level with the optional Infotainment package (JSA) $695 and the Adaptive Cruise Control option (KSG) $495