commaai/openpilot #30208: rawgpsd died during test run

devtekve, 2023-10-09:
### Describe the bug

RawGPS died and it forced me to take control, I don't know how to reproduce it reliably, sorry!

### Provide a route where the issue occurs

e1107f9d04dfb1e2|2023-10-09--21-02-26--0

### openpilot version

0.9.5

devtekve, 2023-10-12:
Similar thing happened on sunnypilot too, I have a stacktrace, this was the issue I created https://github.com/commaai/openpilot/issues/30226 

Stacktrace: 
```
Traceback (most recent call last):
  File "/data/openpilot/openpilot/selfdrive/manager/process.py", line 40, in launcher
    mod.main()
  File "/data/openpilot/system/sensord/rawgps/rawgpsd.py", line 288, in main
    r = setup_quectel(diag)
        ^^^^^^^^^^^^^^^^^^^
  File "/data/openpilot/system/sensord/rawgps/rawgpsd.py", line 223, in setup_quectel
    send_recv(diag, DIAG_SUBSYS_CMD_F, pack('<BHBBIIII',
  File "/data/openpilot/openpilot/system/sensord/rawgps/modemdiag.py", line 65, in send_recv
    opcode, payload = diag.recv()
                      ^^^^^^^^^^^
  File "/data/openpilot/openpilot/system/sensord/rawgps/modemdiag.py", line 48, in recv
    unframed_message = self.hdlc_decapsulate(raw_payload)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/openpilot/openpilot/system/sensord/rawgps/modemdiag.py", line 35, in hdlc_decapsulate
    assert payload[-2:] == pack('<H', ModemDiag.ccitt_crc16(payload[:-2]))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError
```

dimdom69, 2023-10-19:
Been getting this too on OPGM.

adeebshihadeh, 2023-10-22:
I'm seeing 3 recent instances of this, all from your device. Still looking into this, but I suspect we'll have to add some more logging to get to the bottom of this.

devtekve, 2023-10-23:
Could it be then a hardware issue ? 😭 this is where I hate being in Europe 😂 it happens quite often for me

adeebshihadeh, 2023-10-23:
This is likely software.

devtekve, 2023-10-23:
That's reassuring! I'd hate going through the customs process... Anything I can do to help?

dimdom69, 2023-10-23:
I was getting this same error when I was working on a custom feature. I had a made a new daemon that updated once a minute. When I added 'liveLocationKalman' to the SubMaster, I would randomly get this error on startup. I re-implemented the feature using the 'LastGPSPosition' param and the error went away.

devtekve, 2023-10-24:
@dimdom69 on the SunnyPilot failure I was doing something similar, and I was also subscribing to `liveLocationKalman` IIRC. However, on the stock route I didn't make any modification, maybe I was already on some sort of failure state that remained through the reinstall of stock perhaps? 

Also, thanks for hsaring about the `LastGPSPosition` I wasn't aware of it!

adeebshihadeh, 2023-10-24:
> I was getting this same error when I was working on a custom feature. I had a made a new daemon that updated once a minute. When I added 'liveLocationKalman' to the SubMaster, I would randomly get [...]

I don't see how this could cause any issues with rawgpsd. If you have a branch that reproduces this reliably, that would be very helpful.

dimdom69, 2023-10-26:
> I don't see how this could cause any issues with rawgpsd. If you have a branch that reproduces this reliably, that would be very helpful.

Reverted to using `liveLocationKalman` and now I don't get any errors. I guess I somehow got into an error state that resolved itself with a fresh install? Not sure.

devtekve, 2023-10-27:
> Reverted to using `liveLocationKalman` and now I don't get any errors. I guess I somehow got into an error state that resolved itself with a fresh install? Not sure.

I don’t suggest using liveLocationKalman for any realtime gps stuff, it’s better to use gpsLocation and gpsLocationExternal (C3) otherwise the location won’t be updated properly :(

closed 2023-12-08 by adeebshihadeh