commaai/openpilot #30212: navigate on openpilot not shown

ildoonet, 2023-10-10:
### Describe the bug

I use the lastest codes from master branch and enabled experimental features. 

Also I enabled 'map shown left side'.

But it doesn't show any navigate on openpilot related features. 

How should I access it?

### Provide a route where the issue occurs

1

### openpilot version

0.9.5

sshane, 2023-10-10:
Do you have prime? If you tap on the screen while onroad, it will bring up the map. Use connect on your phone to choose a destination.

ildoonet, 2023-10-10:
Thx for the reply

I dont have prime subscription 

Tapping on the screen gives me nothing. 

In comma connect page, I can not see any button for choosing destination. 

Is there any way to connect the device again with qr code? (I cannot find qr code to reconnect)

closed 2023-10-10 by adeebshihadeh

Wolfybox, 2023-10-17:
> Do you have prime? If you tap on the screen while onroad, it will bring up the map. Use connect on your phone to choose a destination.

is a prime necessary if I only want to test navigate on openpilot?