commaai/openpilot #30213: Difference between actual speed and C3X displayed speed.

haram-KONA, 2023-10-10:
### Describe the bug

There is a difference between the vehicle's actual speed and the speed displayed in C3X of approximately 5 km/h.

### Which car does this affect?

Hyundai KONA_EV_2ND_GEN (Korean 2023)

### Provide a route where the issue occurs

1618132d68afc876|2023-10-08--11-59-19--10

### openpilot version

master-ci 0.9.5

sshane, 2023-10-10:
Does your car have the ability to set to imperial (MPH)? Does it match then?

haram-KONA, 2023-10-10:
> Does your car have the ability to set to imperial (MPH)? Does it match then?

yes ~~~
i will test with MPH

haram-KONA, 2023-10-11:
> Does your car have the ability to set to imperial (MPH)? Does it match then?

Oh sorry....
I checked and found that MPH settings are not possible in the vehicle.

sshane, 2023-11-02:
Can you provide a video of the diff so it's easier to find the signal that's correct?

haram-KONA, 2023-11-03:
https://github.com/commaai/openpilot/assets/88036668/d9f7ab67-9dd7-42c0-9744-5b051fd400fc

haram-KONA, 2023-11-03:
Uploading 20231103_110156.mp4…

haram-KONA, 2023-11-03:
difference video

haram-KONA, 2023-11-03:
> Can you provide a video of the diff so it's easier to find the signal that's correct?

this video

https://github.com/commaai/openpilot/assets/88036668/ec10220e-9fd7-4305-99d9-eef94540c424

sshane, 2023-11-03:
Ah we do this for CAN, but not CAN FD cars. Will check it out on our EV6 soon!

haram-KONA, 2023-12-15:
??

royjr, 2024-02-07:
May be similar to https://discord.com/channels/469524606043160576/1152498009427419197/1204814695610056805