commaai/openpilot #30232: Openpilot Long Control Behaves Poorly on 2017-2018 RAV4

dresden196, 2023-10-13:
### Describe the bug

When running chill mode/OP long on a 2017 Toyota RAV4, with a SDSU, and Comma Pedal, OP long occasionally drops to lower gears at highway speeds when not needed to maintain acceleration, will also occasionally stay in lower gears despite matching set speed. I sometimes have to physically step on the gas for OP Long to shift up to a higher gear. Also OP Long will refuse to coast which results in higher MPG usage. Occasionally OP Long will exceed past specified set limit by about 7 mph (Maybe more), only then realizing this, then decrease to specified set limit. 

### Which car does this affect?

Toyota RAV4 2017-2018

### Provide a route where the issue occurs

383ba03aecff2efa|2023-10-08--22-32-36--0

### openpilot version

0.9.5

ntegan1, 2023-10-13:
https://github.com/commaai/openpilot/issues/28047

sshane, 2024-03-20:
Duplicate of https://github.com/commaai/openpilot/issues/28047

closed 2024-03-20 by sshane