commaai/openpilot #30244: Is support available for “Hyundai Genesis GV70 2021”?

cloudran, 2023-10-14:
### Describe the bug

Describe the bug
Is support available for “Hyundai Genesis GV70 2021”?
At comma 3x, the vehicle is not recognized and starts in dashcam mode.
So I tried extracting my car's fingerprint and firmware information and adding them to value.py, but it still goes into dashcam mode.
I am sharing the fingerprint and firmware information I extracted for reference, and if you have any information you need, please let me know.

Which car does this affect?
Hyundai Genesis GV70 2021

Provide a route where the issue occurs
81a6ddf7edcd84f4|2023-10-14--07-45-08

openpilot version
0.9.4

Additional info
Fingerprint information
48: 24, 64: 32, 74: 32, 96: 32, 111: 8, 160: 24, 181: 32, 234: 24, 256: 32, 282: 16, 293: 16, 298: 16, 314: 32, 320: 8, 325: 32, 352: 16, 362: 32, 373: 24, 416: 32, 426: 16, 432: 32, 437: 32, 442: 24, 463: 8, 480: 16, 485: 16, 490: 32, 506: 32, 512: 8, 517: 24, 549: 16, 608: 32, 688: 32, 698: 32, 794: 32, 837: 8, 842: 32, 879: 8, 895: 8, 896: 8, 897: 8, 898: 8, 899: 8, 900: 8, 923: 8, 934: 8, 944: 8, 945: 8, 946: 8, 961: 8, 962: 8, 1040: 8, 1041: 8, 1042: 8, 1043: 8, 1044: 8, 1046: 8, 1047: 8, 1048: 8, 1049: 8, 1050: 8, 1051: 8, 1052: 8, 1057: 8, 1058: 8, 1062: 8, 1063: 8, 1065: 8, 1066: 8, 1070: 8, 1071: 8, 1076: 8, 1090: 8, 1092: 8, 1107: 8, 1141: 8, 1167: 8, 1174: 8, 1187: 8, 1204: 8, 1208: 8, 1209: 8, 1210: 8, 1214: 8, 1215: 8, 1240: 8, 1247: 8, 1264: 8, 1266: 8, 1267: 8, 1278: 8

Vehicle firmware information
{'carParams': {'alternativeExperience': 1,
               'autoResumeSng': True,
               'carFingerprint': 'mock',
               'carFw': [{'address': 1975,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'cornerRadar',
                          'fwVersion': b'\xf1\x00JK1 BCW RR 1.00 , 1.01 (s\x94@ \x03ISH',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1983,
                          'subAddress': 0},
                         {'address': 2000,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00JK1_ SCC -----      1.00 1.02 99110-AR100         ',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 2008,
                          'subAddress': 0},
                         {'address': 1971,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'hvac',
                          'fwVersion': b"\xf1\x00JK1   97255-AR011CONTROL ASS'Y-DATC  1.04JK1 DATC(-)1.5       ",
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1979,
                          'subAddress': 0},
                         {'address': 1988,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00JK1 MFC  AT KOR LHD 1.00 1.04 99211-AR100 210204',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1996,
                          'subAddress': 0}],
               'carName': 'mock',
               'carVin': '00000000000000000',
               'centerToFront': 1.350000023841858,
               'communityFeatureDEPRECATED': False,
               'dashcamOnly': False,
[... 154 lines cut ...]
               'safetyModelPassiveDEPRECATED': 'silent',
               'safetyParamDEPRECATED': 0,
               'startAccel': 0.0,
               'startingAccelRateDEPRECATED': 0.0,
               'startingState': False,
               'steerActuatorDelay': 0.0,
               'steerControlType': 'torque',
               'steerLimitAlert': False,
               'steerLimitTimer': 1.0,
               'steerRateCostDEPRECATED': 0.0,
               'steerRatio': 13.0,
               'steerRatioRear': 0.0,
               'stopAccel': -2.0,
               'stoppingControl': True,
               'stoppingDecelRate': 0.800000011920929,
               'tireStiffnessFactor': 0.0,
               'tireStiffnessFront': 1000000.0,
               'tireStiffnessRear': 1000000.0,
               'transmissionType': 'unknown',
               'vEgoStarting': 0.5,
               'vEgoStopping': 0.5,
               'wheelSpeedFactor': 1.0,
               'wheelbase': 2.700000047683716},
 'logMonoTime': 345529893018,
 'valid': True}

sshane, 2023-10-21:
What harness did you use?

Try this installer URL and engage:

https://installer.comma.ai/commaai/genesis-gv70-2021

cloudran, 2023-10-21:
Thank you very much sshane, but my car doesn't work.
Probably a harness issue.
I purchased a hyundai M harness.
[image]
[image]
This is the connector for my vehicle.
[image]
This is the harness connector purchased from Comma.
[image]
The connector on my car has one less pin, so I took one pin off the harness on the Comma.
But I still get the error.
Anyway, thank you so much for your help.
[image]
[image]
[image]

sshane, 2023-10-22:
According to our docs, Hyundai L is to be used for the 2.5T trim.

cloudran, 2023-10-23:
Thank you very much. I'll get the L type and try again.

sshane, 2023-11-02:
@cloudran how is the new harness?

cloudran, 2023-11-03:
I received my new harness today.
However, my car is out of town, so I won't be able to test it until sometime next weekend. I'll post a review after I test it next weekend.
Thanks.

cloudran, 2023-11-11:
I tested today with a new harness and a new relay box, but I get the same error with the new harness. 
"Harness Relay Malfunction Check Hardware"
I have no idea what is causing this.

sshane, 2023-11-14:
I am seeing the same messages on the same buses. The other harness does not flip the CAN buses, my mistake, it's only a change in shape. You will want to flip the CAN buses to the car as done in this issue: https://github.com/commaai/openpilot/issues/30355

cloudran, 2023-11-18:
https://cdn.discordapp.com/attachments/524611978208215070/1003351781251756182/unknown.png
I followed the link here and tried changing the pin, and when I did, it went into Dashcam mode.
I downloaded the log file with the original harness, it's in the attachment. Thanks.
I changed the file extension bz2 to txt because the bz2 extension file was not uploading.
[81a6ddf7edcd84f4_2023-11-18--18-11-32.txt](https://github.com/commaai/openpilot/files/13400082/81a6ddf7edcd84f4_2023-11-18--18-11-32.txt)
[81a6ddf7edcd84f4_2023-11-18--18-11-40--0--qlog.txt](https://github.com/commaai/openpilot/files/13400104/81a6ddf7edcd84f4_2023-11-18--18-11-40--0--qlog.txt)
[81a6ddf7edcd84f4_2023-11-18--18-11-32.txt](https://github.com/commaai/openpilot/files/13400106/81a6ddf7edcd84f4_2023-11-18--18-11-32.txt)

Lindorak, 2023-11-24:
I’d like to chime in on my issue with my Genesis. 

Car:
Genesis GV70 3.5 Sport 2022 Canada

I have had a “Car Unrecognized” issue since getting my Comma 3X. 

I have fingerprinted my car with no luck. I have followed these instructions for flipping the pins on the harness:
https://github.com/commaai/openpilot/issues/30355#issuecomment-1790990721

Once I did that I still get a “Car Unrecognized”. 

I now don’t get the field of “fwdRadar”. I believe this might be due to the pins being flipped. Although I would get the following line before the flip:
'fwVersion': b'\xf1\x00JK1_ SCC ----- 1.00 1.02 99110-AR100 ',

As seen above, I would get “-----”. I’m sure it is supposed to say “FHCUP” as shown in the values.py. And, as mentioned earlier, after the flip I don’t get that line. So I am wondering if it might be that my car doesn’t have “fwdRadar”?

I have tried to substitute the “fwdRadar” by adding “cornerRadar” line and adding the value to the values.py, but no luck.

sshane, 2023-12-07:
Who here is still having issues? Can you send a latest route and detailed description of what is happening, what you've tried? I would ignore the missing `FHCUP` for now, could be a red herring

cloudran, 2023-12-26:
The problem with my car is that the Korean-made Genesis gv70 was released with HDA2. In the US, the gv70 was released with HDA1. 
Neither sunnypilot nor the original openpilot will work, but the branch made by the Koreans will recognize the car and work.

Lindorak, 2023-12-28:
I have the HDA2 as well since I bought mine in Canada. They discontinued the HDA2 for the HDA1 since chip production couldn't keep up due to the pandemic. Do you know where you got the Korean branch from? I want to review it and see if I can get it working myself.

Lindorak, 2023-12-28:
@cloudran 

Hardware Information:
Dongle ID: 45c39d77edb27ccc

If you are on the official Discord server here is the thread with everything I have done:
https://discord.com/channels/469524606043160576/1175176860003487744

I have tried a few things, the list is below:

1. I have flipped the CAN 0 and the CAN 1.
2. I have tried Sunnypilot.
3. I have tried to fingerprint my Genesis GV70 by forking.
4. I have gone back to the pre-flipped configuration.
5. I am back on stock openpilot, not on my own fork.
6. I have made sure that all the cables are snug and everything looks good. 

Let me know if I missed anything. It's been a while since I played around with my Comma so everything isn't fresh. Thankfully I was documenting everything in the official Discord. 

If you don't have access to the thread I have copy and pasted the entire conversation I've had below:

3:53 PM]Lindorak: Hey everyone! I just got my Comma 3X and had some questions. 

I have a Genesis GV70 (Canadian)

The Comma AI 3X doesn't recognize my car. I followed the fingerprinting guide and I came across something weird. I see that one of the lines has "-----" instead of "FHCUP".

                                                            {'address': 2000,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00JK1_ SCC -----      1.00 1.02 99110-AR100         ',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 2008,
                          'subAddress': 0},

When I keep the "-----" as is I get a plethora of errors in my dashboard in my car. 
[10:19 PM]Lindorak: I made some further changes and tried a few things. So far nothing seems to work 😦
[10:19 PM]Lindorak: Does anyone have a Canadian GV70 and has gotten Comma to work with their car?
[10:20 PM]Lindorak: I have even SSH'd in and double check the code where I made commits and it looks like my changes have been made from my fork.
[10:23 PM]Lindorak: Some more information in case anyone needs it in the future. I have also tried Sunny's repo and it seems I haven't had any luck either 😦
[10:32 PM]kzhawks: Can you share your device with me? (stealing sunnys lingo )
    • Share your device with me (https://connect.comma.ai/, gear icon, share by email, type in github_5355398, share icon again to confirm), set the device name to your year/make/model and your Discord username
[10:32 PM]kzhawks: Also do you have the V4 or V6 version?
[10:47 PM]Shane: You have smart cruise control already, correct?
[10:54 PM]Lindorak: Yes
[10:54 PM]Lindorak: I believe I have HDAII (HDA2, not sure what the correct nomenclature is)
[10:56 PM]Lindorak: I got it before, from what I remember, the great chip shortage. I heard that they stopped supplying the GV70 with the HDA2 for some time. Kinda off topic but just thought I would give you all the details. Kinda off my rocker due to the Adderall I took today.
[10:57 PM]Lindorak: I also double checked the decimal to hexadecimal values and they are correct. 2000 > 7D0 and 1988 > 7C4.
[11:12 PM]kzhawks: this is likely why you are getting " a plethora of errors" when you add the correct fingerprint. I don't think there's been a HDAII GV70 tested yet, if I remember correctly
[11:20 PM]Lindorak: @kzhawks Ah, is that also the reason why I am getting the "-----" instead of the "FHCUP"?
[11:21 PM]Lindorak: And would there be a way to get my GV70 tested or anything I can do myself?
[1:20 AM]Lindorak: Just messaging to see if anyone has any suggestions?

I've read up on Cabana and was wondering if that was the best way to go about trying to add my car to OpenPilot?
[3:57 AM]Shane: @Lindorak what is your dongle id?
[4:02 AM]Shane: can you upload full rlogs for a route?
[4:43 AM]Lindorak: Dongle ID: 45c39d77edb27ccc

I’ll upload the rlog sometime today. Thanks for getting back to me 🙂
[4:47 AM]Lindorak: Route:

https://connect.comma.ai/45c39d77edb27ccc/1700353302151/1700360142125

rlogs waiting for upload to complete
[6:13 AM]Lindorak: I think I did all this correctly. Let me know if I did anything wrong
Attachment file type: unknown
45c39d77edb27ccc_2023-11-18--13-21-41--120--rlog.bz2
[... 114 lines cut ...]
                          'subAddress': 0},
                         {'address': 1971,
                          'brand': 'hyundai',
                          'bus': 1,
                          'ecu': 'hvac',
                          'fwVersion': b"\xf1\x00JK1   97255-AR330CONTROL ASS'Y-DATC  1.04JK1 DATC(-)1.5       ",
                          'logging': True,
                          'obdMultiplexing': False,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1979,
                          'subAddress': 0},
                         {'address': 1988,
                          'brand': 'hyundai',
                          'bus': 1,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00JK1 MFC  AT CAN LHD 1.00 1.04 99211-AR100 210204',
                          'logging': False,
                          'obdMultiplexing': False,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1996,
                          'subAddress': 0}],
[7:28 PM]Lindorak: In this new data dump I don’t see cornerradar
[7:29 PM]Lindorak: Sorry, I meant fwdRadar
[2:45 PM]Lindorak: I have also submitted a comment into the GitHub issue.
[2:56 AM]Lindorak: @Shane should I got back to pre-flip configuration? Or is there potentially another flip method I can do?

cloudran, 2023-12-29:
I used the branch here and recognized the vehicle. Give it a try.
neokii.myds.me
https://github.com/neokii/glidepilot

Lindorak, 2023-12-29:
@cloudran 

I got stuck on:
Registering device - serial (my serial number)
IMEI (my IMEI number)

Did you have trouble with this screen?

Lindorak, 2023-12-29:
Gave the link you tried a shot and no luck :(

Lindorak, 2024-02-16:
@sshane Is there anything you still need from my end to get my Genesis up and running?

wyatt-rematter, 2024-06-05:
@sshane @cloudran @Lindorak I recently installed a c3X on my 2022 Genesis GV70 3.5T with HDAII and am having a similar issue as was described above with the car not being recognized (dashcam mode only) on stock openpilot and when I fingerprinted the vehicle (https://github.com/wyatt-rematter/openpilot) I get the Harness Relay Malfunction issue and all the driver assist systems in the vehicle throw errors. Not sure if any of you made any progress or if this issue went anywhere, but I'd appreciate your input - thank you!

sshane, 2024-06-07:
@cloudran what is that state of your car? Does it all work now?

Lindorak, 2024-06-08:
@wyatt-rematter 

I haven't made any progress since the last time I've posted. Been personally dealing with life so I've put this on the back burner. 

A couple questions:
- Which country are you located in?
- Which trim of the 3.5 do you have? There is the higher trim and the lower trim.