commaai/openpilot #30266: Steering temporarily unavailable

CraigRam, 2023-10-17:
### Describe the bug

I have just installed the comma 3X when the adaptive cruise control is turned on, I get an error message steering temporarily unavailable. I have restarted and reset the device and reinstalled the current version of open pilot. 0.9 .4 with no success. I have checked the compatibility for the vehicle and it is compatible, the Comma team have set up the same, comma 3X on a ram 1500 DT Ltd 

### Which car does this affect?

Ram 1500 DT Ltd 2022 model

### Provide a route where the issue occurs

https://useradmin.comma.ai/?onebox=56f7f07241e11132%7C2023-10-17--10-56-23

### openpilot version

0.9.4

### Additional info

As stated above

sshane, 2023-10-18:
Please try this branch: https://github.com/commaai/openpilot/tree/fix-ram-1500-fault

Or use this installer URL during setup: https://installer.comma.ai/commaai/fix-ram-1500-fault

CraigRam, 2023-10-18:
Thanks sshane that worked. I really appreciate your assistance.

CraigRam, 2023-10-18:
> Please try this branch: https://github.com/commaai/openpilot/tree/fix-ram-1500-fault
> 
> 
> 
> Or use this installer URL during setup: https://installer.comma.ai/commaai/fix-ram-1500-fault

Thanks sshane that worked. Appreciate your assistance.

sshane, 2023-10-18:
Thanks! The EPS torque looks similar, so we're good to merge! You can check out `master-ci` or `nightly` (more stable) branches for now until the next release, replacing `fix-ram-1500-fault` above.

closed 2023-10-18 by sshane

CraigRam, 2023-10-18:
> Thanks for trying the branch! The EPS torque looks similar, so we're good to merge! You can check out `master-ci` or `nightly` (more stable) branches for now until the next release, replacing `fix-ram-1500-fault` above.

Sorry sshane... I am very new to this stuff... What do you mean checkout "master-ci" or "Nightly"... How do i do that?

CraigRam, 2023-10-19:
> Thanks for trying the branch! The EPS torque looks similar, so we're good to merge! You can check out `master-ci` or `nightly` (more stable) branches for now until the next release, replacing `fix-ram-1500-fault` above.

Hi Shane I have tried these branches
https://installer.comma.ai/commaai/nightly
 https://installer.comma.ai/commaai/master-ci

https://github.com/commaai/openpilot/tree/nightly
https://github.com/commaai/openpilot/tree/master-ci

And they don't work I am back to the same error can you please link the original ram fault branch that worked. Cheers
Craig

sshane, 2023-10-19:
I just triggered an the workflow to run so those branches will be updated soon. Please try again in about an hour

CraigRam, 2023-10-19:
> I just triggered the workflow to update those branches. Please try again in about 30 minutes

Just confirming this is the branch to update as still nothing is available 

https://installer.comma.ai/commaai/fix-ram-1500-fault

sshane, 2023-10-19:
That branch has been merged into master, but a more stable branch (updated daily) is `nightly`, so try this now that it's been updated: https://installer.comma.ai/commaai/nightly

CraigRam, 2023-11-04:
Commai miust have updated to 0.9.5 openpilot I am now getting a cruise control fault on car and commai unit. And Cannot use the cruise control system on car.

aboyce64, 2024-05-03:
did this issue ever get fixed i am using the latest version Version 0.9.7 and am getting the steering unavailable issue on my 2019 RAM 1500 Longhorn. Does anyone have a solution for this?

sshane, 2024-05-28:
@aboyce64 what is your dongle id (found in device settings)? It would also be helpful if you could send a route from comma connect: https://connect.comma.ai/

aboyce64, 2024-05-29:
> @aboyce64 what is your dongle id (found in device settings)? It would also be helpful if you could send a route from comma connect: https://connect.comma.ai/

I have found it works just fine using SunnyPilot. But I would like to be able to use stock OP.  I will get the info and come back here and post later.

josephchambers, 2025-09-02:
I am seeing the same/similar issue.  If I start the truck with coma ai usb unplugged and then plug it up once I am driving it works, but if I leave it plugged up and start the truck I get this error.  I've added a USB switch as a work around so I can start up with it powered off then turn it on wich is annoying.  It is on the current gold version.  Which versions is this fix in?