commaai/openpilot #30294: Move UI from GPU -> CPU

adeebshihadeh, 2023-10-20:
part one of making the gpu exclusive for the driving model

jnewb1, 2023-11-17:
@deanlee We are working on larger driving models, but we need to move the UI and a few other components to the CPU for that to be feasible on device. A good start would be analyzing and optimizing the maprenderer. We [moved mapsd to the CPU](https://github.com/commaai/openpilot/pull/30349) as an experiment for CPU rendering, and it takes about 10x the time to render on the CPU (about 30ms, rendering 256x256 at 2hz), so we'd likely need quite a few speedups to render the stylized UI map on device. We have a [nice test](https://github.com/commaai/openpilot/pull/30349) for this in CI that can also run on device. Would you be interested in looking into optimizing that?

deanlee, 2023-11-17:
@jnewb1 will there be any improvement if enable threading?

jnewb1, 2023-11-17:
> @jnewb1 will there be any improvement if enable threading?

The latency would most likely be better with threads (you can try it with LP_NUM_THREADS), however the goal would be trying to fit the whole UI on the cpu without bogging down the system, which would mean we would need to significantly optimize overall, not just use more threads. We also don't really care that much about the latency as long as it's reasonable (mapsd is fine with 30ms delay)

deanlee, 2023-11-27:
Think it's difficult to optimize the performance based on the current available map API (there is also a lack of other header files from mapbox, only `qmapbox.hpp` in `mapbox-gl-native-qt/include`). I've tried using static rending, but always get an error:

>  render error! "Map is currently rendering an image"

I looked at the source code of the qmapbox, but still can't figure out what's causing the issue.
Maybe we can do this optimization later, after switching the map API to MapLibre? https://github.com/commaai/openpilot/issues/26739

adeebshihadeh, 2023-11-27:
Sure, @deanlee wanna try to do the MapLibre move?

andiradulescu, 2024-09-06:
@adeebshihadeh do you prefer this to be done in openpilot, so that we control each process that runs on CPU?

or we want an AGNOS approach, where everything qt and/or EGL runs on CPU without any changes in openpilot?

if you prefer in AGNOS, I will create a separate issue explaining the two possible variants (with weston or without).

adeebshihadeh, 2024-09-06:
There's a tradeoff. More control with openpilot, but then it's a little harder to run arbitrary stuff. I think it'll end up being the cleanest to cut out Weston and do almost everything in openpilot.

closed 2026-11-10 by adeebshihadeh