commaai/openpilot #30302: [Ford Q4] Ford Lightnings Harness Causes Errors When Openpilot Not Running

coffee-cake-isaac, 2023-10-21:
### Describe the bug

There's currently an issue with F150 Lightnings on the Ford Q4 Harness/CAN-FD platform. This is specific to the Lightning as the ICE variants do not have this issue nor do the Mustang Mach Es.

1. Install Ford Q4 Harness into F150 Lightning.
2. Turn on Vehicle and notice no errors
3. Install Comma Power and plug into harness
4. Turn on vehicle and notice no errors
5. Plug in Comma provided USB-C cable to harness (leaving unplugged from Comma 3/3X)
6. Notice the number of errors received (Traction Control, Park Assist, One Pedal Drive)
7. Plug in Comma 3X (with a working release of OpenPilot) and notice issue persist.
8. Allow Openpilot to fully boot and open the relay on the harness and observe the car faults disappearing

### Which car does this affect?

2022+ Ford F150 Lightning

### Provide a route where the issue occurs

N/A - Happens prior to openpilot

### openpilot version

0.9.5

### Additional info

I've spoken with Discord and they're pretty confused on why this is happening as well. Additionally, there's now 2 Lightnings that are observing this behavior -- mine and one other persons.

I've also emailed support and they were a bit confused as well.

adeebshihadeh, 2023-10-22:
Need a route to look into this. Does it also happen with openpilot toggled off in the settings?

sshane, 2023-10-25:
@coffee-cake-isaac do you have a route?

coffee-cake-isaac, 2023-10-25:
Let me quickly grab one. Few seconds.

coffee-cake-isaac, 2023-10-26:
I've responded in PM on discord due to sensitive info on the route. If that's not sufficient, I can get the info away from any sensitive locations.

hiattwl1, 2023-11-07:
Also experiencing this issue, let me know if I can provide any data. I can replicate with the exact steps as above. What really baffles me is the issue occurs without comma connected, like the harness box is immediately engaging relays when cable gets plugged in.

coffee-cake-isaac, 2023-11-08:
Something I tried yesterday that I haven't tried previously was plugging in the USB-C without anything else comma related plugged in. 

So, I plugged in the harness. Then just immediately plugged in an OBD-C cable (provided by comma) and it still faults (even without the Ethernet plugged in from the Comma Power)

adeebshihadeh, 2024-01-02:
$200 bounty for root causing and coming up with a fix we can ship.

coffee-cake-isaac, 2024-01-03:
It's worth noting that JUST the harness with an OBD-C cable is necessary to trigger faults.

So harness with relay box is okay, add the cable only and it fails.

ebfio, 2024-01-12:
Just installed Comma3X with the Ford Q4 harness - same behavior.

rsquared123, 2024-01-15:
Curious if this is bluecruise or copilot assist related. Are both users xlt or lariat/platinum?

garrettpall, 2024-01-16:
Has anyone tried looping pt1 - pt4 from connector to connector and avoiding the harness box. It’s a long shot but depending on what voltages are going through those and how long your obd c cable is there could be some weird electro magnetic stuff going on. At a minimum you would be removing variables from the situation.

coffee-cake-isaac, 2024-01-16:
> Has anyone tried looping pt1 - pt4 from connector to connector and avoiding the harness box. It’s a long shot but depending on what voltages are going through those and how long your obd c cable is [...]

I don't think so. The cable is very short though. Using the one that comes from comma, you're looking at less than a foot of cable length. 

I have taken measurements on some of the pins with a multimeter. I haven't been able to get CAN0H or CAN0L due to where the IPMA is and the length of wiring on the cables.

I was able to measure CAN1 H/L and they are sitting comfortably at 60ohms, which is what DebboR from Discord said it should be reading. However, I have seen CAN2 constantly jumping voltages. It eventually settles at 60ohms for a little bit before jumping all over the place again.

garrettpall, 2024-01-16:
I would make sure the relay driving pins are isolated from any other pins on the harness box while the obd c cable is plugged in. Thinking through it more the pass through arent even on the obd c cable

garrettpall, 2024-01-16:
Having the harness box and obd c cable out of the car and with the cable plugged into the harness box, put one multimeter lead at the end of the obd c cable on SBU1 terminal and make sure all of the pins have on the harness box molex connector have no resistance/are open. Repeat on SBU2 https://github.com/commaai/neo/blob/master/car_harness/OBD-C.sch.pdf

coffee-cake-isaac, 2024-01-16:
> Having the harness box and obd c cable out of the car and with the cable plugged into the harness box, put one multimeter lead at the end of the obd c cable on SBU1 terminal and make sure all of the [...]

Does the OBD-C need to have power from something? Or can I literally just test on my desk for example

rsquared123, 2024-01-25:
Is this a copilot assist or bluecruise vehicle?

coffee-cake-isaac, 2024-01-31:
It's a BlueCruise vehicle.

I have news. Maybe good, maybe bad. I'd loop DebboR but I'm not sure what his GH username is.

So, I've figured out why this was not seen on the external Red Panda. The external Red Panda didn't use a comma power to the harness. It was harness box -> red panda -> Comma 3.

The C3X allowed us to plug the Comma Power into the harness box. I've observed the following pattern:

1) Comma approved OBD-C with no comma power -- no faults.
2) Comma approved OBD-C with comma power -- faults
3) Really long OBD-C cable -- faults.
4) 1m OBD-C cable with no comma power -- no faults

I was able to test a ~6ft USB-C that sent the device to onroad but `controlsd` failed to start. Not sure about that.

What I think I've deduced is there's some kind of necessary shielding that's only possible on higher-end cables. Maybe some form of electromagnetic interference? From something on the vehicle side?

I'm working on getting a longer, higher quality OBD-C cable (maybe going to buy Apple's just for testing purposes). 

Would love to hear if you guys have any suggestions but I do believe this is entirely hardware related and nothing we can ship in OP aside from maybe having the relay always open... which seems not ideal.

I'm not sure if maybe comma wants to offer a longer, specialized OBD-C cable for this vehicle or what the plan would be to support the vehicle.

madsci1016, 2024-01-31:
> The C3X allowed us to plug the Comma Power into the harness box. I've observed the following pattern:
>
> 1. Comma approved OBD-C with no comma power -- no faults.
> 2. Comma approved OBD-C with comma [...]

Not a Comma or openpilot user, but my friend is that had this issue. I am an Electrical Engineer. Based on a brief read at what it seems Comma is doing here, which is sending the vehicles CAN bus down a long USB-C cable. This is either adding a unterminated leg to the CAN bus STAR configuration, assuming it already a star (which is not recommended in theory) or turning their flat bus into a star by adding that long leg and it's unterminated with the camera disconnected. 

This would cause single reflections on the CAN bus disrupting all the systems people are reporting. Remove the long leg of the star (created by the USDC cable) reflections stop. Use a shorter cable, the reflections are less intense. 

This would also explain the behavior you stated, assuming the Comma design doesn't have static termination but does so through some sort of CAN transceiver that has to be enabled by it's boot process. 

This could be proved by creating a termination plug to go onto the end of the long USB-C cable when the camera is disconnected. It would have to be the USB connector and a 120 ohm resistor across the CAN data lines. I can't seems to find a pinout of the UCBC cable as wired by openpilot to see whats pins that would be though.

coffee-cake-isaac, 2024-01-31:
> > The C3X allowed us to plug the Comma Power into the harness box. I've observed the following pattern:
> > 
> > 1. Comma approved OBD-C with no comma power -- no faults.
> > 2. Comma approved OBD-C with comma power -- faults
> > 3. Really long OBD-C cable -- faults.
> > 4. 1m OBD-C cable with no comma power -- no faults
> 
> Not a Comma or openpilot user, but my friend is that had this issue. I am an Electrical Engineer. Based on a brief read at what it seems Comma is doing here, which is sending the vehicles CAN bus down a long USB-C cable. This is either adding a unterminated leg to the CAN bus STAR configuration, assuming it already a star (which is not recommended in theory) or turning their flat bus into a star by adding that long leg and it's unterminated with the camera disconnected.
> 
> This would cause single reflections on the CAN bus disrupting all the systems people are reporting. Remove the long leg of the star (created by the USB-C cable) reflections stop. Use a shorter cable, the reflections are less intense.
> 
> This would also explain the behavior you stated, assuming the Comma design doesn't have static termination but does so through some sort of CAN transceiver that has to be enabled by it's boot process.
> 
> This could be proved by creating a termination plug to go onto the end of the long USB-C cable when the camera is disconnected. It would have to be the USB connector and a 120 ohm resistor across the CAN data lines. I can't seems to find a pinout of the USB-C cable as wired by openpilot to see whats pins that would be though. Maybe someone can chime in on that?
> 
> As to why only the lightning? Well, EVs have much simpler CAN networks, usually only point to point or very few nodes while ICE vehicles are likely already a STAR configuration with many nodes, and as far as other EVs, well the Lightning is big and probably has the longest run of CAN bus already vs a MachE, which makes it more susceptible to reflection interference.

My smooth brain is having trouble following any of this.

https://github.com/commaai/neo/blob/master/car_harness/OBD-C.sch.pdf Does this help though?

alan-polk, 2024-01-31:
> > My smooth brain is having trouble following any of this.
> > https://github.com/commaai/neo/blob/master/car_harness/OBD-C.sch.pdf Does this help though?
> 
> The final missing piece is how the relay box/harness is wired, as this would tell me what CAN busses are still passed through down the USB cable when the relay is "closed".
> 
> But worst case it's all, so a proper test would be to make a USB C female port (so the cable you have can plug into it at the camera side) and 4x 120 ohm resistors across the following pins:
> 
> A10-A11 A2-A3 B2-B3 B10-B11
> 
> If plugging in this "termination plug" in the problem cable fixes the issues with the vehicle, it proves my theory, though sadly i don't think there's a quick fix to make this mechanism act as a "Failsafe" as intended, because the camera coming unplugged can't magically make this shorting plug become connected, though maybe a passthrough shorting plug can be made, as it shouldn't hurt the Comma camera is the CAN buses are double terminated at the camera side, or even better be configured to not terminate the bus internally.

Agreed.  I am looking at making a USB-C coupler with two female USB-C breakout boards for testing, I can add 120 ohm resistors to the coupler.

hiattwl1, 2024-01-31:
> > > My smooth brain is having trouble following any of this.
> 
> > > https://github.com/commaai/neo/blob/master/car_harness/OBD-C.sch.pdf Does this help though?
> 
> > 
> 
> > The final missing piece is how the relay box/harness is wired, as this would tell me what CAN busses are still passed through down the USB cable when the relay is "closed".
> 
> > 
> 
> > But worst case it's all, so a proper test would be to make a USB C female port (so the cable you have can plug into it at the camera side) and 4x 120 ohm resistors across the following pins:
> 
> > 
> 
> > A10-A11 A2-A3 B2-B3 B10-B11
> 
> > 
> 
> > If plugging in this "termination plug" in the problem cable fixes the issues with the vehicle, it proves my theory, though sadly i don't think there's a quick fix to make this mechanism act as a "Failsafe" as intended, because the camera coming unplugged can't magically make this shorting plug become connected, though maybe a passthrough shorting plug can be made, as it shouldn't hurt the Comma camera is the CAN buses are double terminated at the camera side, or even better be configured to not terminate the bus internally.
> 
> 
> 
> Agreed.  I am looking at making a USB-C coupler out of two female USB-C breakout  boards to support testing, I can add 120 ohm resistors to the coupler.

Ditto. Just ordered two breakout boards. 

I made the comment in discord about the relays in the harness box not disconnecting all can lines. From what little information there is in the schematic, I believe the relay only disconnects can0 and can2 then they send can1 and can3 down as a star. 

I'm wondering if the harness box needs a second set of relays for that set of lines. 

Two high level contributors brushed it off as a non hardware issue. If comma isn't using fail safe relays on all can bus communication across their "OBD-C" cable design, then that's a failure in safety in my opinion.

unhappyelf, 2024-01-31:
> The final missing piece is how the relay box/harness is wired, as this would tell me what CAN busses are still passed through down the USB cable when the relay is "closed".
>
> But worst case it's all, [...]

Im a computer engineer and I support this message.

madsci1016, 2024-01-31:
> I'm wondering if the harness box needs a second set of relays for that set of lines.
>
> Two high level contributors brushed it off as a non hardware issue. If comma isn't using fail safe relays on all [...]

As someone who's has designed CAN bus systems before, I'm actually shocked this is how Openpilot was designed. I assumed looking at what my buddy told me, the "relay box" (which i never saw) was a box that had a MCU in it itself, to handle all the CAN bus tapping and "middle-man attacking" and it's logical controller just talked to the camera over the USB cable as a proper USB connection. 

When i realized it's just some wire splices that add all these super long stubs to the CAN bus itself, all sorts of hairs on my neck started to tingle. Lots of design rules and approaches violated on something i would never cut any corners on (A motor vehicles communication bus).

alan-polk, 2024-01-31:
> As someone who's has designed CAN bus systems before, I'm actually shocked this is how Openpilot was designed. I assumed looking at what my buddy told me, the "relay box" (which i never saw) was a [...]

Probably relevant to note that Comma's designed was built around the IPMA being behind the rear view mirror and the Comma on the windshield.  Everything was 6 inches away from each other.  It wasn't until Ford came along and put the IPMA under the dash (or in the trunk for some SUVs) that created issues with cable length.

madsci1016, 2024-01-31:
> Probably relevant to note that Comma's design was built around the IPMA being behind the rear view mirror and the Comma on the windshield. Everything was 6 inches away from each other. It wasn't until Ford came along and put the IPMA under the dash (or in the trunk for some SUVs) that created issues with cable length.

Thanks, that makes more sense. 
Basic rules for CAN bus design disallow Stub lengths longer than a critical length governed by bus speed. Even if terminated. and thats 1.7m and 3m for 500kbit and 250kbit respectively. 

These USB cables with can bus signals... seem to be that long or worse.

alan-polk, 2024-01-31:
> Thanks, that makes more sense. Basic rules for CAN bus design disallow Stub lengths longer than a critical length governed by bus speed. Even if terminated. and thats 1.7m and 3m for 500kbit and [...]

It does seem that if long distances between the can device and the comma are going to become common, then the system needs a redesign so that the communications between the harness box and the Comma are true USB-C, and the USB to CAN conversion happens at the harness box located right at the CAN device.

madsci1016, 2024-02-02:
My buddy sent me a pic of the relay module. It's only a DPDT relay, and it seems to have the termination resistors (3 of them) on board. So i have to guess there's no termination resistors on the camera? 

Since the relay is DPDT, there's no way the schematic is anything other than CAN 0,1,2 are ALWAYS connected to the USBC cable, and the relay just shorts them when off. So you have 3 unterminated 1-2m stubs hanging off the one CAN bus of the car between the main computer and the car's camera/radar. That's a very bad design for CAN and very likely to be the reason for these issues, as the stubs creating signal reflections are pumping interference on the CAN bus.

jyoung8607, 2024-02-02:
> My buddy sent me a pic of the relay module. It's only a DPDT relay, and it seems to have the termination resistors (3 of them) on board. So i have to guess there's no termination resistors on the camera?

Manufacturer designs vary. A CAN bus wants about 60 ohms total termination resistance in the form of 120 ohm termination resistors at each end of the bus. Sometimes (often) the camera is the far end of the bus and supplies termination resistance, sometimes it isn't and there's already 60 ohms between the CAN pair with the camera disconnected. There's optional jumpers in the harness design to configure it for either variant. We need to check if the Ford harness design is set up properly to match the camera.

https://github.com/commaai/neo/blob/master/car_harness/open_pinout.sch.pdf
(see the lower left corner for jumper configurations)

coffee-cake-isaac, 2024-02-02:
>We need to check if the Ford harness design is set up properly to match the camera.
> 
> https://github.com/commaai/neo/blob/master/car_harness/open_pinout.sch.pdf (see the lower left corner for jumper configurations)

I'd imagine it would be since it's working for other CAN-FD Ford users (Mach E, ICE F150)

jyoung8607, 2024-02-02:
> I'd imagine it would be since it's working for other CAN-FD Ford users (Mach E, ICE F150)

It needs to be verified anyway. Partial missing bus termination is something you'd get away with *sometimes*.

The Ford Q4 harness is configured for single-ended termination, with the camera as a stub. P23-P25 are jumpered to provide 60 ohms toward the camera and P3-P5 are left open which means we assume 60 ohms already present on the vehicle side. Has anyone actually unplugged the factory camera and measured resistance between the CAN bus pins?

https://github.com/commaai/neo/blob/master/car_harness/Ford_Q4_Harness.sch.pdf

All that said, while I don't have a schematic of the harness relay box, I'm not sure the termination settings come into play until the relay opens. @madsci1016 has valid points about max stub length and the OBD-C cable does become part of the bus whether the relay is open or closed. It's actually going to form TWO stubs (bus 0 and 2 are electrically bridged with the relay closed) which won't help matters at all.

alan-polk, 2024-02-02:
> It needs to be verified anyway. Partial missing bus termination is something you'd get away with _sometimes_.
>
> The Ford Q4 harness is configured for single-ended termination, with the camera as a [...]

I'm assuming you mean unplug the IPMA and not the camera.  On the Q4 Fords the camera is an optical cable that runs to the IPMA.

If I unplug the connector from the female side of the harness (this should be the car side of the harness) These are your resistances:

CAN1H to CAN1L: 122 ohm
CAN2H to CAN2L: 62.6 ohm

If I leave the connector from the car to the harness box unplugged and also unplug the harness box, the back side of the harness box connector should be the actual IPMA (It's impossible to get to the actual IPMA plug itself without removing it from under the dash).  These are your resistances:

CAN0H to CAN0L: open
CAN1H to CAN1L (Pins 8 and 10): open
CAN1H to CAN1L (Pins 19 and 20): 122 ohm
CAN2H to CAN2L: 49.34 k-ohm

This was all with the truck off.  Let me know if I need to repeat with truck on.

alan-polk, 2024-02-05:
> It needs to be verified anyway. Partial missing bus termination is something you'd get away with _sometimes_.
>
> The Ford Q4 harness is configured for single-ended termination, with the camera as a [...]

Jason

I've been waiting on your feedback before making a custom coupler with additional resistors.  Any advice after seeing the resistance measures I pulled from the truck?

BennyTheBeaver, 2024-03-18:
Any updates on the progress? 

I'm really hoping to see the Lightning be supported, only thing holding up my purchase. :-)

DallasSehlhorst, 2024-03-19:
I don't know if mentioned previously, but on my Lightning I only get faults when the Comma supplied usb-c cable is plugged in (without the Comma 3X attached or turned on).  I have a usb-c extension cable attached to the obd box, and once I plug the Comma usb-c into it faults appear.  In other words: with ONLY the usb-c extension cable plugged in no faults appear, so I don't think this is a length or interference issue.  I don't have another cable to test that will plug into the recess of the 3X to test if the supplied cable is truly the issue.

alan-polk, 2024-03-19:
Isaac was able to confirm yesterday that the harness I modified for him resolves the issue. I clipped all of the pass through wires from the harness box and looped them all back at the two IPMA plugs, then I extended ignition, ground, can1, and can2 by 9 feet so that the harness box can be relocated to the rearview mirror and the stock ODB-C cable can be utilitized. I put 60ohm resistors on the camera side harness between can1H/L and can2H/L.  I don't know if a hardware solution qualifies for the bounty, but it resolves the issue.

[image]
[image]

hiattwl1, 2024-03-20:
> Isaac was able to confirm yesterday that the harness I modified for him resolves the issue. I clipped all of the pass through wires from the harness box and looped them all back at the two IPMA [...]
Thank you!!!!!

Does anyone have a reliable source for the three connectors needed to make this harness? The data I'm finding is several years old. 

I want to make a new one vs hacking up my current cable. 

I would like the ford IPMA connectors as well as the part for the harness box and all pins if they don't come with them already. 

I hope this path will be cheaper than another $200 harness. Maybe comma can do right by us and provide some custom cables like this?

alan-polk, 2024-03-20:
I was only able to find the female half available.  For the male half, the only places that sell them require a minimum order of 300 of them.  Perhaps Comma would be willing to sell use a few loose ones.  The other problem is they use some proprietary pin settings tools which I couldn't find, and I had a lot of trouble inserting the pins cleanly.  I've rewired 5 restored cars in my life, I've got a lot of automotive wiring experience, and I couldn't get the connectors to cleanly seat and pass through.  Hence why I eventually hacked one up.

https://www.mouser.com/ProductDetail/TE-Connectivity/1924955-6?qs=M6jHmRuQorVt%252Bccjv9Htyg%3D%3D&countryCode=US&currencyCode=USD

https://www.mouser.com/ProductDetail/TE-Connectivity-AMP/2288276-1?qs=iR2ablhfrmFQ0tikRKv%252B8w%3D%3D&countryCode=US&currencyCode=USD

https://www.mouser.com/ProductDetail/Molex/501646-2600?qs=AcmJtBVhzkR3B7PmD3EBJQ%3D%3D&countryCode=US&currencyCode=USD

https://www.mouser.com/ProductDetail/Molex/501647-1100?qs=MeCAwI9ZxERfLG%252BPYEOBVw%3D%3D&countryCode=US&currencyCode=USD

https://www.te.com/usa-en/product-1438981-1.html

windydrew, 2024-03-26:
I have a 23 Lightning Lariat with my newly installed x3. I have the same issues as everyone else and willing to test whatever to help out.

gustavoclimaco, 2024-04-01:
Hello guys, can you confirm if the bad of the CAN  is the same as my results from the CABANA (CAN 2 = NA)? I'm doing an adaptation on a Ford Fusion 2017 (with lane centering adapted by changing modules) with Harness Q3 so I have no comparison  parameters, other than your error.
[image]

coffee-cake-isaac, 2024-04-10:
I've been running a long harness that @alan-polk has put together with 0 issues and 0 faults. Near stock look when installed as well. 
@adeebshihadeh Does a hardware approach satisfy the issue? Is that something comma would be willing to handle or not because it's a separate one off?

alan-polk, 2024-04-11:
The Ford CANFD IPMA is physically too large to fit in the headliner/rearview mirror area, it will always have to be under the dash or in the trunk. A 6" Q4 harness is doing nobody any good.  It seems that a longer Q4 harness solves a multitude of problems around USB cable length and canbus deflections.  If a Q4 harness were offered in two lengths, it should cover every current and conceivable future CANFD vehicle from Ford.

8-Foot Q4 harness: 21+ F150, 21+ Lightning, 22+ Expedition, 24+ Ranger, 23+ Super Duty (if encryption can be solved), 24+ Mustang (if encryption can be solved), Future Bronco 6g (mid-cycle refresh will have CANFD). 
16-foot Q4 harness: 21+ MachE, 23+ Escape, Future Bronco Sport, Future Edge, Future Explorer

thenoname, 2024-04-11:
> Isaac was able to confirm yesterday that the harness I modified for him resolves the issue. I clipped all of the pass through wires from the harness box and looped them all back at the two IPMA plugs, then I extended ignition, ground, can1, and can2 by 9 feet so that the harness box can be relocated to the rearview mirror and the stock ODB-C cable can be utilitized. I put 60ohm resistors on the camera side harness between can1H/L and can2H/L. I don't know if a hardware solution qualifies for the bounty, but it resolves the issue.
> 
> ![IMG_1404](https://private-user-images.githubusercontent.com/56047663/314258259-b0f6a3bb-d6ce-4c44-8ad8-718e18718835.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI4MDk1NTMsIm5iZiI6MTcxMjgwOTI1MywicGF0aCI6Ii81NjA0NzY2My8zMTQyNTgyNTktYjBmNmEzYmItZDZjZS00YzQ0LThhZDgtNzE4ZTE4NzE4ODM1LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDA0MjA1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1Y2Y1OWY2OGJmODQyZDQwZDVkYzA5MzcxMGRjYzI3MWMxZWY3ODc0NDgyMzg4MjE1MTlhOTkxY2NkZTU1OWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.kYCTbDpYNQWcyo9NTDHk3KQkh3LGFlB4ZnBBRP_lImk) ![IMG_1403](https://private-user-images.githubusercontent.com/56047663/314258267-5fc9eed2-d26b-4c09-a4fc-4734e6355e0e.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI4MDk1NTMsIm5iZiI6MTcxMjgwOTI1MywicGF0aCI6Ii81NjA0NzY2My8zMTQyNTgyNjctNWZjOWVlZDItZDI2Yi00YzA5LWE0ZmMtNDczNGU2MzU1ZTBlLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDA0MjA1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc3NDQ3ZTJjMmZmYjBlZjA5NjJlNmQ5M2QwZjVkMWJlNTY2YTNmMzY0NWE2NmQ4ODIxYThhZDhmNmNlNzE2NGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.O7hqnAMuWt3w3phTiBsEHL52rrrqCZDr9UZg3njk52g)

Would you be able to draw a schematic for the changes you made? I would like to try them out myself

Mjack007, 2024-04-13:
Installed Comma3X with the Ford Q4 harness in my 2023 XLT ER Lightning no Blue Cruise. Getting these errors at startup then resolve after Comma 3x is fully booted up.
[image]

alexose, 2024-04-29:
Finally got to modifying my harness.  Works great now!  No errors at all.

I wrote up a [visual guide](https://docs.google.com/document/d/1keSikuriLE7qTHo2GoivIsX3x6P-m_xfNeW5oRVqzzY/edit) of the process for others looking to do the same thing.

alan-polk, 2024-04-29:
I was able to create an extension harness so that no cutting of the Comma harness was necessary.  

Shopping list:

(2) White Male Plugs:
https://www.mouser.com/ProductDetail/Molex/501646-2600?qs=AcmJtBVhzkR3B7PmD3EBJQ%3D%3D&countryCode=US&currencyCode=USD

(1) Black Female Receptacle
https://www.mouser.com/ProductDetail/Molex/503091-2621?qs=OlC7AqGiEDkAIGQFvezNCg%3D%3D&countryCode=US&currencyCode=USD

(32) Pins
https://www.mouser.com/ProductDetail/Molex/501647-1100?qs=MeCAwI9ZxERfLG%252BPYEOBVw%3D%3D&countryCode=US&currencyCode=USD

3D printable Coupler Enclosure
https://www.thingiverse.com/thing:6600129

Crimpers:
https://www.amazon.com/gp/product/B078WPT5M1/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

Resistors:
https://www.amazon.com/gp/product/B07QDXTH1V/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

Heat Shrink:
https://www.amazon.com/Wirefy-180-Heat-Shrink-Tubing/dp/B084GDLSCK/ref=sr_1_1_sspa?crid=1K7UD8AJFZOS&dib=eyJ2IjoiMSJ9.uKZEOjO5jb_qXjEFmeryEb4he7IchWPuE1HsgQalSHEelAte1iul13W4yHKk7XLzt4dZvi6jumAPyIT0HCi2x7kaeavJtidyFOMtHpTRPfCFb3R_CfRR80bVa3EgA7LFuVcQzy0cu7EUGng3FVq_rNoy9Fk1Kz-eKI784vik0_v_6bps5GdOd-7v2PrcbJ-XR9sKPqigXyBo1PBZEJmfbn2nW0v4HIiWISRQC-f16Mw.kW-NOe02_ilyb96_NHuXW12sPopxYDVSIKBY-xoYjxs&dib_tag=se&keywords=1mm+heat+shrink+tubing&qid=1714362061&sprefix=1mm+heat+shrink+tubing%2Caps%2C107&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1

DIY VIdeo:
https://youtu.be/6-FBSqa1O9E

windydrew, 2025-02-25:
Extending my harness fixed my issue as of 2 months ago. 0 faults to date. I am currently making extension harnesses for a few other users as well.

ebfio, 2025-02-25:
> Extending my harness fixed my issue as of 2 months ago. 0 faults to date. I am currently making extension harnesses for a few other users as well.

I'd love one if you still have availability, @windydrew.

LowCutKilt, 2025-03-31:
I'd also be interested in one @windydrew 😀

alan-polk, 2025-05-12:
> All the harnesses are here: https://github.com/commaai/neo/tree/master/car_harness

We have managed to prototype a custom USB-C Coupler with 120-ohm resistors integrated. This resolves the issues and only requires the USB-Coupler to be swapped from the current solution shipped.

I have a schematic for the custom USB-C coupler, but have not gotten a PCB design or quote yet.  Since Comma already has custom PCBs for their other products, is this something you are interested in taking over?

alan-polk, 2025-06-30:
> Just to confirm, the ICE F150 variants have no issues?

confirmed

madsci1016, 2025-07-01:
> $200 bounty for root causing and coming up with a fix we can ship.

So was this a real offer? Asking since... well...

> This could be proved by creating a termination plug to go onto the end of the long USB-C cable when the camera is disconnected. It would have to be the USB connector and a 120 ohm resistor across the CAN data lines.

> We have managed to prototype a custom USB-C Coupler with 120-ohm resistors integrated. This resolves the issues and only requires the USB-Coupler to be swapped from the current solution shipped.

I was responsible for identifying the issue and suggesting the (now proven) fix...

alan-polk, 2025-07-01:
> So was this a real offer? Asking since... well...
>
>
>
> I was responsible for identifying the issue and suggesting the (now proven) fix...

Shouldn’t the bounty go to the guy who spent hundreds at PCBWay getting the prototypes made?

thenoname, 2025-07-01:
@madsci1016 I have to say, as an outsider, you are acting a pretty rude to someone who really put a lot of effort into this to help the community...

madsci1016, 2025-07-01:
> The story you referenced doesn't work in this scenario. In the story, there wasn't a third person who bought the hammer for the "expert" only for that "expert" to claim all the profit for themselves. If so, I think the moral would be quite different.

No, Alan watched me hit the spot, and decided to start selling hammers to the other people that owned the same boats with the same issue. He should be made whole by the sales of those hammers, while i gave up my knowledge (which costs me hundreds of thousands of dollars in my various college degrees for me to identify the cause of an issue connecting a camera to a car of which neither have i ever touched) for free, and the only recourse for me is that bounty. 

Story hits quite perfectly, and I'm sorry if your missed nuisance of it made you think i was trying to be rude

> Effort is important but experience and knowing where to direct that effort makes all the difference.

alan-polk, 2025-07-01:
> > The story you referenced doesn't work in this scenario. In the story, there wasn't a third person who bought the hammer for the "expert" only for that "expert" to claim all the profit for themselves. If so, I think the moral would be quite different.
> 
> No, Alan watched me hit the spot, and decided to start selling hammers to the other people that owned the same boats with the same issue. He should be made whole by the sales of those hammers, while i gave up my knowledge on where to use them for free to the whole community, and the only recourse for me is that bounty.
> 
> Or to put it simply "Shouldn't that go to the guy that spent $100k on college degrees to know where to tap that hammer (or plug in that resistor?)
> 
> Story hits quite perfectly, and I'm sorry if your missed nuisance of it made you think i was trying to be rude. I'll post the last line of it again in case you missed it:
> 
> > Effort is important but experience and knowing where to direct that effort makes all the difference.
> 
> I never said i was against splitting it or discussing it to some degree, but it is kinda annoying (on principle) this bounty was never marked as solved and payment discussed by either party involved with fixing it. As said i don't have a comma or a lightning, so i went as far as i could to outline the problem and what the fix should be (to literally stick a resistor on a USB plug). I gained nothing personally in the few hours i spent digging through the designs and documentation of a project i had no prior knowledge of and writing these detailed posts on what the problem and likely solution is the betterment of that project and community itself.

A) I'm not talking about me. Comma would never give me a bounty; I'm persona non grata to those guys. And while your idea was key to the final solution, Anthony spent dozens of hours learning to design a PCB and hundreds of dollars at PCBWay for a prototype.  

B) Your assumption that he would be made whole  by selling the devices he designed is flawed, because if Comma pays out the bounty and ships the devices with every Lighting kit... then who would buy one from Anthony since they got one free with the Comma?

madsci1016, 2025-07-01:
> A) I'm not talking about me. Comma would never give me a bounty; I'm persona non grata to those guys. And while your idea was key to the final solution, Anthony spent dozens of hours learning to [...]

Before I walk away as “Let’s agree to disagree” I want to take one last shot at summarized the series of events to see if anyone can point out where I’m incorrect to see if I’m missing something that would otherwise explain the feelings and editorialization going on here by all parties. 

1. A bounty is offered “for root causing and coming up with a fix we can ship.”
2. I am the first to explain the very sophisticated physics of the electrical phenomenon that is the root cause, describe a very simple fix (that could be shipped in a variety of ways) with literal pinouts ( so short of drawing a literal picture or schematic, but doing the same in words). Which is just “USB port with resistors”
3. I do not go into the widget selling business with a USB port with resistors since I don’t have knowledge of the goals or structure of this project to understand which final approach most aligns with their vision. (personally, I’d rather see the interface box redesigned to not pass the CAN buses up the USB cable) 
4. Another user with little to no experience in designing PCBs follows my pinout/described design and spends hundreds of dollars (due to his inexperience) making prototype revisions of a board with USB ports and resistors with no real plan to sell them to make that money back because:
5. That approach probably doesn’t align with the goals, structure or vision of the project which you are saying would be to officially add the adapter to a Comma kit? (and not draw anything from Anthony’s input?) The exact reason I didn’t do it?
6. But a few here think Anthony should now get the bug bounty anyway since “he spent money” and I only spent my time? Though only one of us did the "root causing and coming up with a fix we can ship"?

Is there significantly more to Anthony’s PCB design than a USB passthrough and resistors? Was there more he added to it electrical engineering wise? Is that what I’m missing? Otherwise, I feel this is a bit of a highjack in the grand scheme of things. I laid out a design and why it works, someone else ran with it and now he should get the bounty reward money?  

 I learned to design PCBs the way Anthony just did (but decades ago). I want to be clear that it is great to do so. I still have a tackle box full of easily $1000s in design revisions of things i never sold (so for me only) or never made my money back if i did sell them. I have a long history in OSHW and OSS. I’ve run had my own OSHW hardware designs stolen and sold lower than I was offering in forum group buys (I’m that old) before the community even voted on the OSHW logo. Yeah this time I didn’t spend the 10 minutes to make a .sch/brd file but did put in the larger work of creating and describing the design and how it fixes the problem. And this somehow feels a bit like my design getting stolen, again. Unless I’m missing something.

madsci1016, 2025-07-01:
> 
> You have clearly shown your vastly superior intellect, tenacity, and passion to the rest of us. I hope Comma pays you your bounty.
> 
> As for Anthony, anyone who wants to help offset the cost he incurred trying to help the community, please contacrt me on the lightning forum.

Ahh well an ad-hominem reply answers my question i guess. Like i said earlier i was never against a discussion on the reward payment split, i brought this up on principle. OSS/OSHW the primary payment is usually credit and attribution for sharing designs and improving the community, which i did by empowering Lightning users to fix this issue, no? Yet the bounty stays open and i do not see my handle mentioned anywhere on the Lightning forum. Thats the principle that bothered me.

Uncle-Tony, 2025-07-16:
> > You have clearly shown your vastly superior intellect, tenacity, and passion to the rest of us. I hope Comma pays you your bounty.
> > As for Anthony, anyone who wants to help offset the cost he incurred trying to help the community, please contacrt me on the lightning forum.
> 
> Ahh well an ad-hominem reply answers my question i guess. Like i said earlier i was never against a discussion on the reward payment split, i brought this up on principle. OSS/OSHW the primary payment is usually credit and attribution for sharing designs and improving the community, which i did by empowering Lightning users to fix this issue, no? Yet 18 months later the bounty stays open, no real attribution here (in comments or in the act of closing the bounty) and i do not see my handle mentioned anywhere on the Lightning forum. Thats the principle that bothered me, as vain as it sounds. But vanity drives OSS since money doesn't.
> 
> Anyway, i feel i had to get succinct to defend my POV when my illustrative example seem to fail and that succinctness came off as arrogant as a result for that i am sorry. I'm truly glad a cheap solution was possible and that a capable user was able to arrange forum purchases to get that solution to the geeky users in need. I'm told now by a few acquittances that use it (including the friend with a lightning 4 states way that originally asked me to look at and fix the issue 18 months ago) maybe there's some disfunction in this project that would easily explain why this bug bounty is left open and that i shouldn't take it as a slight? (And maybe why Alan is persona non grata?)

It sounds like you don't actually understand why the solution works and haven't kept up with the project after multiple people have worked together to actually understand it fully and actually fix the issue. A termination as you suggested works, but puts strain on the CAN modules as it causes a pretty severe over termination. A "Tuning" resistor over the proper bus resolves the reflections caused by a stub that exists in the lightning without causing potential damage to peoples vehicles. If you feel like you deserve the bounty because of a half ass suggestion your threw out without testing at all, I can send you the $200, regardless if I get to claim the bounty or not.

madsci1016, 2025-07-17:
> It sounds like you don't actually understand why the solution works and haven't kept up with the project after multiple people have worked together to actually understand it fully and actually fix the issue. A termination as you suggested works, but puts strain on the CAN modules as it causes a pretty severe over termination. A "Tuning" resistor over the proper bus resolves the reflections caused by a stub that exists in the lightning without causing potential damage to peoples vehicles. If you feel like you deserve the bounty because of a half ass theory you threw out without testing at all, I can send you the $200, regardless if I get to claim the bounty or not. And thank you Alan, but I am happy just being able to get this to work properly in the lightning. I love my truck a lot more and am happy to share that with everyone else.

The editorialization is getting offensive. 

3 months this issue was open, discussion hadn't gotten anywhere except "maybe get better cables". 

I wrote out **784** words explaining the phenomenon at play, suggesting several avenues to implement changes to fix, and even mentioned the fact a passthrough plug would be over-terminating and might need to be corrected in some manor. Not to mention the document search and reviewing i had to do to better explain the issue and provide pinouts for the fix/testing for those capable of doing so. 

784 words is not "half ass theory I threw out" or "simply suggested adding resistors". It's was the very foundation that enabled you to  "understand the issue". Is your board doing anything other than adding termination resistors of some value to the can bus to prevent reflections from the stub? Value changed due to over termination? Is Alan's harness doing anything other than preventing the stub from existing? Congrats, not only do I fully understand it, i spelled it all out for you almost two years ago. 

Geez. My comments here have been referenced in discord discussion for fixing a similar issue in Riviana apparently. 

Yeah i stopped there, i didn't have a lightning or a comma to do any testing myself, and i didn't have personal time to try and work with someone 4 states away. 

It was never about the money, i said that. It was about to recognition for  "root causing and coming up with a fix" which i thought i did, as several implemented fixes including yours have proved the issue was as i described. 

The money was just a mechanism for the recognition, since none of you ever even bothered to take a few seconds to thank me for starting this discussion you followed. I guess the behavior to not close bounties by the devs is just a reflection on how the community itself behaves anyway.

Next time i know not to bother trying to help a community I'm not a part of because they will never treat you as anything other than an outsider.