commaai/openpilot #30303: Device crashing with beeping from car harness

j0dan, 2023-10-23:
### Describe the bug

My Comma 3 started crashing a few months ago. It will turn off without warning, and the harness will start beeping while it boots up again.

Two times I've seen a 'camera malfunction' error, and sometimes, the system will not restart without power cycling it.

Most of the crashes happened on forks, but I've been running the official one for a few weeks to capture some there. However, it doesn't seem to record/upload some of the drives where it's happened.

I have nothing in the realdata/crash folders, and I don't see anything else in /var/log related to the crash.

The route attached here seems to begin right after the crash and nothing prior to the crash is in useradmin.

### Provide a route where the issue occurs

d64e45c407898145|2023-10-19--18-42-32--0

### openpilot version

master (unknown what I was on last Thursday)

Euclid71, 2023-11-02:
I had the same problem with two different comma3 devices in the same car.  I replaced the harness with no change, however, when I replaced the OBD-C cable, the problem went away.

sshane, 2024-03-30:
@j0dan does this still occur?

j0dan, 2024-03-30:
Not presently. If I had to guess it's heat based, and summer is over. Thanks for following up.

closed 2024-05-17 by adeebshihadeh