commaai/openpilot #30310: Offline selfdrive test improvements

sshane, 2023-10-23:
Run: `pytest -n auto --dist=loadscope -o cpp_files=test_* -m 'not slow'`

- [ ] Some tests fail:
  ```python
  FAILED common/tests/test_swaglog::swaglog
  FAILED common/tests/test_util::util::read_file - xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 44, column 19
  FAILED common/tests/test_util::util::safe_fwrite - xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 45, column 46
  FAILED common/tests/test_common::swaglog
  FAILED system/loggerd/tests/test_loggerd.py::TestLoggerd::test_bootlog - openpilot.common.timeout.TimeoutException: Timed out after 5 seconds
  FAILED system/proclogd/tests/test_proclog::buildProcLogerMessage
  FAILED system/loggerd/tests/test_loggerd.py::TestLoggerd::test_init_data_values - openpilot.common.timeout.TimeoutException: Timed out after 5 seconds
  ```
- [ ] Changes local state (either modifies or wipes params, have to re-auth each time)

jnewb1, 2023-10-30:
@sshane can you make sure you are up to date on poetry install? I think the xml error will be parsed correctly in the latest version of pytest-cpp https://github.com/pytest-dev/pytest-cpp/pull/117

sshane, 2023-12-01:
test_util and test_ratekeeper were old files that were later consolidated to test_common, they pass now once removed.

Last two failures are from loggerd:

```
FAILED system/loggerd/tests/test_loggerd.py::TestLoggerd::test_bootlog - openpilot.common.timeout.TimeoutException: Timed out after 5 seconds
FAILED system/loggerd/tests/test_loggerd.py::TestLoggerd::test_init_data_values - openpilot.common.timeout.TimeoutException: Timed out after 5 seconds
```

These are from `_gen_bootlog` function, `./system/loggerd/bootlog` takes longer than 5 seconds, as well as crashes on setting the log since my system has been up 70 days and the log is too long

https://github.com/commaai/openpilot/blob/e687be939e7cf67c4a75b87320ab058941d4316f/system/loggerd/bootlog.cc#L44

adeebshihadeh, 2023-12-17:
I think we can close this now? `pytest .` works reliably for me in the root.

closed 2023-12-17 by adeebshihadeh

adhintz, 2024-02-04:
I'm getting the same two timeout failures as @sshane does above when `_gen_bootlog` running `./bootlog`

adeebshihadeh, 2024-02-04:
> I'm getting the same two timeout failures as @sshane does above when `_gen_bootlog` runs `./bootlog`

I suspect this isn't a test bug, but rather something the bootlog waiting on something your machine doesn't have (NVMe?). Can you repro by just running `systemd/logged/bootlog`? If so, let's make a separate issue and we can look into it

adhintz, 2024-02-04:
Sounds good, will do. Do you know what params I should specify when running
it?

adeebshihadeh, 2024-02-04:
You shouldn't need to specify anything and it should finish in ~1s.

sshane, 2024-02-04:
If your machine has been up for months it could take longer than a second.

adhintz, 2024-02-05:
Good call. The reason the test is failing is because `create_params_path` in `params.cc` cannot create the params directory:
```
$ system/loggerd/bootlog 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to ensure params path, errno=13, path=/home/user/.comma/params, param_prefix=/d
Aborted (core dumped)
```
My params directory is owned by a user id 100999 which appears to be something that docker created. chowning it should fix the issue.

The timeout hides the underlying directory permission issue. For some reason when `test_loggerd.py` runs the failing bootlog with `subprocess.check_output("./bootlog"` it takes 14s to run. When running the failing bootlog directly in a terminal it takes < 1s. I'm not sure what's causing the huge difference.