commaai/openpilot #30313: I want to add a vehicle
Hyundai 2023 Staria
발음듣기복사하기

2nahyung, 2023-10-24:
(Ecu.fwdRadar, 0x7D0, None): [
b'\xf1\x00US4_ RDR ----- 1.00 1.00 99110-CG000 ',
],
(Ecu.cornerRadar, 0x7B7, None): [
b'\xf1\x00US4 BCW RR 2.00 , 2.00 (xT\x97#\x01S\x06\x14',
],
(Ecu.fwdCamera, 0x7C4, None): [
b'\xf1\x00US4 MFC AT KOR LHD 1.00 1.06 99211-CG000 230524',
],
(Ecu.hvac, 0x7B3, None): [
b'\xf1\x00US4 97255-CG160CONTROL UNIT 1.02.00US4DATC(-)0.9 ',
],

sshane, 2023-10-25:
What is your dongle id? Can you open a pull request with a car port? Generally, Hyundai car ports are rather simple

sunnyhaibin, 2023-10-26:
> What is your dongle id? Can you open a pull request with a car port? Generally, Hyundai car ports are rather simple, as long as it has ACC and LKAS.
> 
> Closing if we get no response as this isn't actionable without more info.

This is being actively worked on. Can close this issue for now.

closed 2023-10-26 by sshane

itkairos2023, 2023-11-28:
> > What is your dongle id? Can you open a pull request with a car port? Generally, Hyundai car ports are rather simple, as long as it has ACC and LKAS.
> > Closing if we get no response as this isn't actionable without more info.
> 
> This is being actively worked on. Can close this issue for now.

I placed an order to install comma 3x on my Straa vehicle. When my product arrives, I want to share the data.