commaai/openpilot #30341: Use both pitch and roll angles in steering compensation

simontheflutist, 2023-10-26:
The roll compensation formula in the [0.8.13 blog post](https://blog.comma.ai/0813release/) is

$$
\dot v = \ldots + g \theta
$$

where $\theta$ is the car's roll angle (treated via a small-angle approximation). Considering that `locationd` provides both roll $\theta$ and pitch $\phi$, I would propose to replace this formula with

$$
\dot v = \ldots + g \cos \phi \sin \theta.
$$

Linearization shows that $\cos \phi \sin \theta = \theta + o(|\phi| + |\theta|)$ for small angles $\phi$ and $\theta$, but it occurs to me that including second-order effects might help with further reducing the steering error that remains.

simontheflutist, 2023-10-26:
Here is a Python function to that compares my proposed compensation to the existing compensation in terms of [grade](https://en.wikipedia.org/wiki/Grade_(slope)#Example_slopes_comparing_the_notations) and [cross slope](https://en.wikipedia.org/wiki/Cross_slope).

```
import numpy as np

def kinematic_orientation_compensation(cross_slope, grade):
    sin_theta = cross_slope / (1. + cross_slope**2)**0.5
    cos_phi = 1. / (1. + grade**2)**0.5
    
    # incumbent compensation factor
    theta = np.arcsin(sin_theta)
    # new compensation factor
    a = sin_theta * cos_phi
    
    print(f"linear compensation error is {(theta-a)/a * 100:.2f}%")
    print(f"which is {(theta - a) * 9.8:.2f} m/s/s")
```

For a highly contrived example, let's combine the following values:
- up to 10% superelevation in [WSDOT guidelines](https://wsdot.wa.gov/publications/manuals/fulltext/m22-01/1250.pdf)
- 23% grade on a [street in SF](https://en.wikipedia.org/wiki/Grade_(slope)#Roads)

```
>>> kinematic_orientation_compensation(cross_slope=0.1, grade=0.23)
linear compensation error is 2.78%
which is 0.026 m/s/s
```

Verylukyguy, 2023-10-28:
> If this acceleration went uncorrected for, say, 10 seconds, the horizontal drift would be 1.3 meters. Of course, the openpilot control loop (including the planner) would be able to soak up much of the error in due time. But it is always better not to overwork the integral controller.

It has been widely noted that on long smooth curves OpenPilot seems to "give up" at some point and if you touch the steering wheel for a second, OpenPIlot "resets" and completes the curve on its own.

A couple years ago, @JasonJShuler saw this and said that is was likely a bug buried deep in OpenPilot from the beginning.
I wonder if this is it.

simontheflutist, 2023-10-28:
@twilsonco latest [patch](https://github.com/simontheflutist/openpilot/commit/a94c8165a73b279e523ddba10680e527ddb3f706#diff-548081f0795b50d035f9846bbdfa1f8c3ea2331717dda4624470e30131bbb795R55) is a better implementation than feeding straight into control. This plugs into the existing orientation-for-control filtering strategy in `paramsd`, avoiding difficult questions of pitch noise.

simontheflutist, 2023-10-28:
> It has been widely noted that on long smooth curves OpenPilot seems to "give up" at some point and if you touch the steering wheel for a second, OpenPIlot "resets" and completes the curve on its own.

What I am concerned about is the implementation of live parameter estimation in paramsd and torqued (e.g. [observability issue?](https://github.com/commaai/openpilot/blob/master/selfdrive/locationd/paramsd.py#L90)) Adaptive control is difficult to do right, and history is full of examples such as the MIT rule where a very reasonable controller failed in unexpected ways.

Verylukyguy, 2023-10-28:
> @twilsonco latest [patch](https://github.com/simontheflutist/openpilot/commit/a94c8165a73b279e523ddba10680e527ddb3f706#diff-548081f0795b50d035f9846bbdfa1f8c3ea2331717dda4624470e30131bbb795R55) is a [...]

With you guys working together, I can't wait to see the results.

nuwandavek, 2023-11-06:
@simontheflutist this is conceptually in the correct direction. The bulk of the work, like with roll compensation, is in making sure the pitch value from the localizer can be completely trusted, on a large dataset. Roll from the localizer had a failure (from various causes) of once every thousand minutes or so, which had to be validated and addressed. I'm assuming it is the same for pitch. A few things like tapping the device I think cause pitch spikes, which need to be validated too. I'm adding this to my list of things to do, but may not get to it soon.

simontheflutist, 2023-11-06:
But is the pitch significantly worse than roll? If not, a [`paramsd` feedforward](https://github.com/simontheflutist/openpilot/commit/593a5698788f4969d3de58649da17c09d3215b2f) would seem to benefit from the noise attenuation inherent to parameter kf.

nuwandavek, 2023-11-07:
It may not be significantly worse than roll (this needs to be verified though). But there are a few ways in which roll is different from pitch. For example, the device does not move/shake the same way in roll-direction than in pitch direction, when it is tapped, or when the car goes over potholes. This can lead to some sketchy and subtle lateral movement. But all of these things can be verified and handled.
If you're using this in your fork at the moment, I'd suggest you look at the above things. Also, you want to use the `calibratedOrietationNED` since there is non trivial pitch difference due to mounting, that we adjust for.

nuwandavek, 2023-11-08:
Not a priority at the moment. Will revisit later when appropriate.

closed 2023-11-08 by nuwandavek