commaai/openpilot #30355: Hyundai Tucson L 2023（china） can error

1okko, 2023-11-01:
### Describe the bug

After startup, openpilot prompts CAN ERROR and the car dashboard displays LKAS fault

### Which car does this affect?

Hyundai Tucson L 2023(CHINA，1.5T 8AT )

### Provide a route where the issue occurs

8ffe737851ca1b11|2023-11-01--01-26-25--0

### openpilot version

0.9.4

1okko, 2023-11-01:
[image]
[image]

sshane, 2023-11-01:
Looks like you have the wrong harness, the messages exist on bus 2 (camera side):

[image]

Which harness did you buy?

sshane, 2023-11-01:
openpilot reports everything is fine with that route if we flip the buses in the logs, so I recommend trying to flip the CAN bus on the harness. There should be users in Discord to help with that, or you can view the wiring diagram for your harness in the above link.

1okko, 2023-11-01:
When I haven't connected to C3X yet, everything is normal, indicating that the wiring harness should be fine，The model I am using is Hyundai N

1okko, 2023-11-01:
> openpilot reports everything is fine with that route if we flip the buses in the logs, so I recommend trying to flip the CAN bus on the harness. There should be users in Discord to help with that, or you can view the wiring diagram for your harness in the above link. You just take certain wires out and flip them.

HOW to flip?

1okko, 2023-11-01:
> openpilot reports everything is fine with that route if we flip the buses in the logs, so I recommend trying to flip the CAN bus on the harness. There should be users in Discord to help with that, [...]
fliped CAN2 AND CAN0 ,Problem solved，thank you

sunnyhaibin, 2023-11-02:
> HOW to flip?

Instructions can be found under Description -> harness type of this PR:
- https://github.com/commaai/openpilot/pull/27392

sshane, 2023-11-02:
Great! Unfortunately we don't have any harness that is the opposite of this, @vanillagorillaa prefers not to add another variant just yet until we see more cars that require this harness. Feel free to check back in the future

closed 2023-11-02 by sshane