commaai/openpilot #30357: Black screen on boot in low temperature

lskov, 2023-11-01:
### Describe the bug

In low temperature the Comma3 doesn’t turn on the screen during boot. 
Need to heat up the C3 display before it will boot with screen initialized. 

Similar to previous bug [here](https://github.com/commaai/openpilot/issues/27514). And it worked for some time, but seeing it again now with cold weather.
But it’s still able to operate the car and it’s working well, it’s just the screen there doesn’t turn on.

Temperature where approx 10degC the other morning in the route below.

### Provide a route where the issue occurs

7291d51974eb2d1d|2023-10-31--07-15-54

### openpilot version

0.9.5

adeebshihadeh, 2023-11-01:
@robbederks got any ideas or want to try an RMA?

robbederks, 2023-11-01:
Think we probably need to RMA

adeebshihadeh, 2023-11-01:
@lskov open a ticket at https://comma.ai/support and post a screenshot of this conversation. we'll swap the device for you and see if that fixes it. once your screen is here, we'll also test it to see if we can also reproduce it

closed 2023-11-01 by adeebshihadeh