commaai/openpilot #30362: NVMe drive not mounted alert after driving

goodchoom, 2023-11-02:
### Describe the bug

Twice so far, I have encountered the NVMe drive not mounted alert after driving and going offroad. The last time this occurred was on September 28th.

In both cases, I have not noticed an impact in performance, also the alert disappeared after the very next drive. However the route prior to the alert never showed up in connect, presumably as a result of the NVMe drive not being mounted at the time.

I have never disassembled my comma device for any reason; it has almost always stayed plugged in and mounted on my car’s windshield.

dongle ID: 5c49984c8efb9c9c

### Provide a route where the issue occurs

N/A

### openpilot version

60e585a51dd5c5f9852c9f553217a52ae254562e

morrislee, 2023-11-03:
i have had this issue, after needing to reseat it multiple times. the last time when I did it, I worked the drive up and down a bit before I fully push it down, havent had that message since

goodchoom, 2023-11-07:
Unfortunately, this issue is now occurring more frequently. A reboot will temporarily resolve it but will resurface after a drive or two. Many of my drives aren't being recorded as a result.

morrislee, 2023-11-10:
where abouts are you located? If you are near me, I dont mind reseating your NVMe drive for you. Its very easy. I have taken my C3 apart a few dozen times

adeebshihadeh, 2023-11-10:
Unfortunately, this is likely a HW issue with the device. Open a ticket at comma.ai/support, reference this ticket, and I'll make sure to get it taken care of.

closed 2023-11-10 by adeebshihadeh