commaai/openpilot #30404: Comma 3 boots into Dashcam Mode

Verylukyguy, 2023-11-07:
### Describe the bug

The Comma 3 boots up intermittently in Dashcam Mode, seemingly after restarting the vehicle and resolves if the vehicle is restarted again.
It is fingerprinting properly using FPv1;  there are no missing Platforms on my Useradmin page.

### Provide a route where the issue occurs

df51b2195de66a9b|2023-11-03--18-17-10--0

### openpilot version

Comma Nightly @ 105ffa0

sshane, 2023-11-08:
When you added your fingerprint in https://github.com/commaai/openpilot/pull/21545, it used a script which did not check the origin bus, as usually 0/2 is the same bus (PT and camera), but on ASCM GM, 2 is the chassis. There happens to be the same address with different lengths on 0 and 2 (like ECMEngineStatus), so some addresses in your FP have an incorrect length on bus 0.

Verylukyguy, 2023-11-08:
There were also earlier attempts to add my vehicle's fingerprint: https://github.com/commaai/openpilot/pull/2440 and https://github.com/commaai/openpilot/pull/21030

Verylukyguy, 2024-04-08:
It does not appear to be a Fingerprinting issues because the Platform is correct on my Useradmin page when this happens.
This issue devolved after a change in OpenPilot after a commit was merged.

Verylukyguy, 2024-04-08:
df51b2195de66a9b/0000005a--78b136e680/0

sshane, 2024-04-08:
I should have a fix next week for you to test.

Verylukyguy, 2024-04-15:
I am ready when you are @sshane.

Verylukyguy, 2024-04-27:
Here is another route:  df51b2195de66a9b/00000001--bc22ed8054/0

Verylukyguy, 2024-06-08:
Here is another route:  df51b2195de66a9b/00000007--12eddfcd96/0