commaai/openpilot #30436: Stop trying to accelerating when traction is lost.

jakethesnake420, 2023-11-10:
When driving on slippery roads and traction is lost, openpilot continues to apply the throttle. Openpilot should detect traction is lost either by monitoring the wheel acceleration or through the cars internal traction control system and back off the throttle.

ErichMoraga, 2023-11-11:
https://discord.com/channels/469524606043160576/524328425415245827/702194645953806386

George said, "we disengage on traction control" ~2.5 years ago.  Has that changed?

haraschax, 2023-11-12:
Issues need logs, please follow the issue format. We support hundreds of cars.

closed 2023-11-12 by haraschax

jakethesnake420, 2023-11-13:
Sure it's easy to disengage but the point of the issue is to prevent the traction control from engaging in the first place. Human drivers accelerate slower when it's slippery.