commaai/openpilot #30446: No Panda Issue

tglevrie, 2023-11-12:
### Describe the bug

My Comma 3x turns on when starting the vehicle, but says "No Panda." When I was switching over to the 3X from the 3, I accidentally disconnected a grey cable from the harness, which I was afraid might affect it. However, it didn't seem to effect anything (noticable). It seemed to work just fine for a week or two, but now doesn't work at all. Could the harness be why it's not reading the panda?

### Provide a route where the issue occurs

1c05783affe6641f|2023-01-20--19-06-44

### openpilot version

0.9.4

### Additional info

I uploaded the most recent video I had, but I'm unsure if the issue was displayed in the video.

tglevrie, 2023-11-12:
I've repaired the harness wire, but it's still not functioning.

adeebshihadeh, 2023-11-12:
Can you try the release3-staging branch? You can install it with this URL: openpilot-test.comma.ai.

tglevrie, 2023-11-12:
Hey, that worked!

closed 2023-11-12 by adeebshihadeh