commaai/openpilot #30455: Unable to run unit tests

nworb-cire, 2023-11-14:
### Describe the bug

I am unable to run panda unit tests on a fresh clone of master. This appears to be related to 2ad82cbfb064676401876a20e49e3965c40344a0.

Steps to reproduce:
```bash
cd /data
rm -rf openpilot scons_cache
git clone --filter=blob:none --recurse-submodules https://github.com/commaai/openpilot.git
cd openpilot
scons -j$(nproc)
python3 -m pytest panda/tests/safety

ERROR: usage: __main__.py [options] [file_or_dir] [file_or_dir] [...]
__main__.py: error: unrecognized arguments: -n --dist=loadscope panda/tests/safety
  inifile: /data/openpilot/pyproject.toml
  rootdir: /data/openpilot
```

### OS Version

AGNOS 8.2

sshane, 2023-11-14:
@jnewb1 is this expected to work on device?

jnewb1, 2023-11-14:
agnos doesn't have xdist installed (until https://github.com/commaai/agnos-builder/pull/181 and next agnos release), so it won't work on device. CI uses the test.sh script within tests/safety/, which is what I would recommend to run the panda tests.

closed 2023-11-14 by jnewb1

adeebshihadeh, 2023-11-14:
Generally, this is expected to work, but we didn't want to delay some pytest refactors on an AGNOS update. For now, you can `pip install` the new packages on your device.