commaai/openpilot #30487: Camera Fails to Start After Snoozing Update Notification

borgmon, 2023-11-18:
### Describe the bug

### Description:
Users are experiencing a lock-up where the camera fails to start after snoozing an update alert. This issue occurs under a specific set of circumstances where the device has not been connected to the internet for an extended period.

### Steps to Reproduce:
1. Do not connect the Comma device to the internet for 2 weeks.
2. Start the car, triggering an update alert from the Comma device.
3. Click "Snooze" on the update alert.
4. Drive the car for some time and then stop.
5. Wait for 2 minutes and start the car again.
6. The update alert shows up again; click "Snooze".
7. The camera fails to start, and this issue persists even after rebooting from the menu.
8. Disconnecting and reconnecting the device after a few minutes temporarily resolves the issue.

### Expected Behavior:
After snoozing the update alert, all functions, including the camera, should continue to work as normal until the user decides to perform the update.

### Actual Behavior:
The camera fails to start after the user snoozes the update alert multiple times.

### Suggested Fixes:

1. **Bug Fix**: Investigate the reason behind the camera lock-up post snooze action. It could be a software bug that fails to reinitialize the camera service after the update prompt is snoozed.

2. **UX Improvement**: Change the snooze functionality to update the "last checked time". This way, users won't be prompted for an update every time the car starts, but rather at the intended intervals (e.g., every other week).

3. **Notification Improvement**: Implement a notification system that reminds users to back up their recordings before connecting to the internet and performing an update. This ensures users are aware of the need to preserve their data. And make users have more incentive to upload their data. 

4. **Update Policy Review**: Re-evaluate the update policy to make it less intrusive for users who may not have frequent internet access. This could involve allowing longer periods between mandatory updates. 

5. **Communication Transparency**: Ensure that update prompts accurately reflect the availability of an update to avoid user confusion and mistrust.

### Provide a route where the issue occurs

N/A

### openpilot version

0.9.4

adeebshihadeh, 2023-11-18:
Do you have a route with this issue?

borgmon, 2023-11-18:
Hi! I don’t have a route ID since there isn’t a route. 
After pressing the snooze, it stays on the dashboard UI without launching the camera.

adeebshihadeh, 2023-11-18:
What's your dongle ID? We'll at least need an approximate date/time this happened to look into it.

sshane, 2023-12-07:
Got a dongle id for us @borgmon?

borgmon, 2023-12-13:
6786a6de3345d600. I have an update on how to reproduce the issue: If you put your car to "back" before the cam started, then you get to skip the first snooze, however it will still be broken next time start the car.

adeebshihadeh, 2024-01-05:
I haven't been able to reproduce this at all. The snooze button is working as intended.

Given that and the dongle ID you posted isn't real, I'm gonna close this. If you've got real logs of this or more complete repro steps, we can look into this again.

closed 2024-01-05 by adeebshihadeh