commaai/openpilot #30495: Low speed steering issue is back with 22' VW Arteon (US model).

dukeajuke, 2023-11-19:
### Describe the bug

I've been using Openpilot for about a year now, and originally run into a problem when I first started using it. I submitted a bug report, and user/dev jyoung8607 fix it in a custom branch for me. That branch was eventually merged into the following release. After the last release, the issue has come back. When the system tries to steer at very low speeds I get "Steering temporarily unavailable" shows up and steering is disabled until I restart the car. See on both these drives:

c34ef43981b6d888|2023-11-19--12-31-55--0
c34ef43981b6d888|2023-11-19--11-52-54--0

### Which car does this affect?

VW Arteon 2022 (US model)

### Provide a route where the issue occurs

c34ef43981b6d888|2023-11-19--12-31-55--0

### openpilot version

openpilot 0.9.4

### Additional info

See bug description

sshane, 2023-11-20:
Do you have a link to the fix?

dukeajuke, 2023-11-25:
Sorry I looked but can't find it. I stopped using that branch when the fix was merged into the master.

jyoung8607, 2023-11-27:
He's referring to #26694. It's specific to a new generation of EPS. I think it may be an angle-rate limit, still investigating.

liltimtim, 2024-01-11:
Also having the same issue. 2018 VW Atlas SEL 

https://connect.comma.ai/9eab6bb0c76baa3d/1704842420631/1704843640843

Was at stop light and accelerated and turned under OP and my control. I felt a slight bump then steering became unavailable.

Gyrohammer, 2024-01-19:
I'm experiencing similar symptoms: 2024 Jetta SE

[33f21a08ace10dde|2024-01-18--17-24-49--1](https://connect.comma.ai/33f21a08ace10dde/1705616689305/1705617351378)

In highway traffic car came to a stop, went along for a bit, then lost steering control/communication. Occurs at the end of the drive.

jyoung8607, 2024-01-27:
Current thinking is the [effective default minSteerSpeed](https://github.com/commaai/openpilot/blob/eada51ec8ed570a4efe70909760460c976cd4f04/selfdrive/controls/lib/latcontrol.py#L6) is right on the edge of what the rack accepts. @Gyrohammer has been testing with a `minSteerSpeed` of 1.0 m/s for a few days with positive results. He's now testing with 0.5 m/s.

Gyrohammer, 2024-02-01:
> He's now testing with 0.5 m/s.

I've been testing at this limit for the past six days with promising results. I have not had this fault occur at-all; everything has been running as expected!

jyoung8607, 2024-02-03:
> I've been testing at this limit for the past six days with promising results. I have not had this fault occur at-all; everything has been running as expected!

Can we have you give it one more try at 0.4 m/s? If possible, I want to still show VW with a zero min steer speed in the official docs, but 0.5 m/s is just a shade over 1 mph. Thank you so much for your help testing so far!

Gyrohammer, 2024-02-04:
> Can we have you give it one more try at 0.4 m/s? 

Will do! I'll update this post in about a weeks time.

Gyrohammer, 2024-02-12:
> > Can we have you give it one more try at 0.4 m/s?
> 
> Will do! I'll update this post in about a weeks time.

Well its been about a week, I've had little to no issue with this modification. I was using it for about 10min in heavy stop and go traffic with zero issue as well, no issues.

jyoung8607, 2024-02-14:
@dukeajuke There's a candidate fix in #31450. At your convenience, try installing it and see what you get!

@Gyrohammer If you've been seeing low speed takeover alerts during your tests, this fix should make stopping silent again.

To test, uninstall openpilot and reinstall from custom URL https://installer.comma.ai/jyoung8607/vw-min-speed

dukeajuke, 2024-02-14:
Thanks! I'll try it tonight.

PR #31450 (2024-05-08): VW: raise minimum steering speed to fix a fault
**Description**

Resolves #30495.

**Fix**

Slightly increased Volkswagen default `minSteerSpeed`, from the [effective global default of 0.3 m/s](https://github.com/commaai/openpilot/blob/eada51ec8ed570a4efe70909760460c976cd4f04/selfdrive/controls/lib/latcontrol.py#L6) up to 0.4 m/s. Since we already considered 0.67 mph as "steer to zero", tweaked the doc generator to show the same for VW's 0.89 mph.

We could definitely do this only for the affected EPS rack-families, but we won't get EPS firmware for installs without comma power, and getting ba [... cut]
closed 2024-05-08 by sshane