commaai/openpilot #30498: Comma 3 blank screen on boot for 0.9.4 and 0.9.5 (but fine for 0.9.3 and below)

philosophydan, 2023-11-19:
### Describe the bug

Short version:
1. Flashed AGNOS to the Comma 3. 
2. Entered openpilot.comma.ai as the custom software source. 
3. Now the comma 3 no longer shows anything on the screen (only if I install openpilot 0.9.4 or 0.9.5 -- older software still works fine)

Further info: 
- When I unplug it (for 30 mins) and plug it back in, it shows the comma logo for <.5 seconds, then goes black. 
- Blue light on the back keeps flashing, but nothing on the screen after quick flash of boot logo. 
- I've tried flashing AGNOS three times (that works fine). That allows me to go through the software install, but after the "operating system" update, the screen goes blank, and no amount of letting it sit unplugged makes it come back on again (except for the very fast flash of the comma logo when it's discharged for a while).

### Provide a route where the issue occurs

N/A

### openpilot version

0.9.5

philosophydan, 2023-11-19:
Dongle ID: 84b95a98fbcb3b7a

adeebshihadeh, 2023-11-20:
This is likely a HW issue that's now manifesting because we check the board ID in the 0.9.4 XBL to detect comma three or 3X. Open a ticket at comma.ai/support and we'll get this taken care of.

closed 2023-11-20 by adeebshihadeh