commaai/openpilot #30506: Mazda: 2024 CX-5 Signature OEM Front Camera Sensor Failure Error

DouglasSherk, 2023-11-21:
### Describe the bug

This appears to be exactly the same as #25525 but on a 2024 CX-5 Signature. Copying and pasting that report as it's still exactly accurate:

> When driving with OP active, front camera sensor error shows up on the OEM instrument cluster
> 
> similar error to https://vf.vin/static/uploads/images/sr2/mazda/2022/02/18/2018-mazda-cx-5-jm3kfbdm5j0335098-img-13.jpg
> 
> Error goes away with disengage/rengage OP
> 
> No errors seen on C3 unit and self drive does not disengage and does not seem affected

I have tried on:
* Stock OP 0.9.5
* A custom version of 0.9.5 I made with @Jafaral's [mazda-tja](https://github.com/Jafaral/openpilot/tree/mazda-tja) commits rebased onto it: https://github.com/DouglasSherk/openpilot/tree/mazda-tja

### Which car does this affect?

2024 Mazda CX-5 Signature

### Provide a route where the issue occurs

60ed2cf2b490292c|2023-11-19--01-58-03--5

### openpilot version

0.9.5

### Additional info

I noted that:

> The lane lines are no longer present on the lane-keeping assist dashboard, so I'm guessing it has something to do with that. When I unplug the C3 but keep the harness in-place, everything goes back to working stock, so it's not a problem with the wiring.

@Jafaral suggested the following:

> Lane lines going away is just a result of the camera error, and not the cause of it., most of the info you see on the dashboard are just decorations really.
> CAN data might reveal more info

Hence I collected the route and filed this report.

I'm a total newbie at OP, but a competent developer otherwise if that's helpful.

sshane, 2023-11-21:
- Can you take a picture of the exact errors you're getting?
- Nothing displayed on the C3/3X and it behaves normally?
- At any specific times do the errors show up?

DouglasSherk, 2023-11-21:
I went back to stock OP 0.9.5, recorded a new route, and captured a video of the error in action. This was as I had OP active and approached a fully stopped car.

https://github.com/commaai/openpilot/assets/1310914/e95b183c-e70d-4cff-a08f-d628bb346a82

Answering your questions:

> Can you take a picture of the exact errors you're getting?

It's "Front Camera Sensor System Fault"

> Nothing displayed on the C3/3X and it behaves normally?

Aside from a "steering temporarily unavailable" error that I had patched over but re-appeared in stock 0.9.5, C3 works exactly as expected. That error doesn't appear to functionally affect anything.

> At any specific times do the errors show up?

It seems to be more frequent when OP is active and I'm approaching another car.

> If you suspect it's caused by one of our messages not matching stock, you can replace `values` with `cam_msg` in mazdacan.py -> create_alert_command function (not update any values) and see if it persists

Thanks, I'll try that soon.

sshane, 2023-11-22:
Also, try to change the min steering speed to something like 5 mph, I suspect that it might vary per trim or model.

DouglasSherk, 2023-11-22:
> Also, try to change the min steering speed to something like 5 mph, I suspect that it might vary per trim or model. Do you notice it fail to steer near a stop/when this happens?

The camera malfunction appears while closing in on the car ahead at around 10 mph or lower. The steering error happens when the car comes to a full stop. Despite the steering error, I haven't noticed any issues with steering. I believe @Jafaral patches over this warning in one of his forks here: https://github.com/Jafaral/openpilot/commit/59c9c8d33fd7a89e366a3f9f1866f72a3aae2495

DouglasSherk, 2023-11-22:
> Also try to set `CAM_LKAS->LINE_NOT_VISIBLE` = 1, and generally match the behavior of the camera's signals. For example `CAM_LANEINFO->TJA` changes values when coming to a stop as well

The first part I think is clear: you're suggesting changing this line https://github.com/Jafaral/openpilot/blob/devel/selfdrive/car/mazda/mazdacan.py#L13 to:
```
lnv = 1
```

The second part seems to go beyond my current understanding. I think you're suggesting something like what's happening in this fork? https://github.com/DouglasSherk/openpilot/blob/mazda-tja/selfdrive/car/mazda/mazdacan.py#L97

Is there a way I can see those values across time in my routes?

sshane, 2023-11-22:
Yes, we have a tool called Cabana built for this: https://github.com/commaai/openpilot/tree/master/tools/cabana. It's exceptioanlly useful for stuff like this!

[image]

sshane, 2023-12-07:
@DouglasSherk did you find anything useful to fix the faults?

siddchou, 2023-12-28:
Any update on issue

DouglasSherk, 2023-12-28:
Someone on the OP Discord mentioned that sunnypilot's quiet mode skips many of the annoying beeps and alerts. That's the biggest problem for me, so I'm going to try that, and if it works, I probably won't bother going any further. The "front camera sensor failure" error is annoying, but it doesn't get in the way.

sshane, 2024-03-20:
@DouglasSherk does this still happen?