commaai/openpilot #30507: 2024 CX-5 Signature not starting automatically from stop

DouglasSherk, 2023-11-21:
### Describe the bug

Potentially related to #30506, or if not, its route works for this issue too.

Mazda Radar Cruise Control ("MRCC)" and/or Traffic Jam Assist ("TJA") is/are designed such that, if the car comes to a full stop for longer than 3 seconds in traffic, then you need to either tap the gas or press the MRCC RES (resume) button before the car starts driving again.

@MoreTore told me that stock OP's implementation is supposed to automatically resume with no action by emulating RES button presses as per @Jafaral's previous work. So it seems it's not behaving as intended.

### Which car does this affect?

2024 CX-5 Signature

### Provide a route where the issue occurs

60ed2cf2b490292c|2023-11-20--15-26-24--2

### openpilot version

0.9.5

sshane, 2023-11-21:
I do not see any examples in that route of you engaged at a stop, and you are on a custom fork with TJA changes. Can you try on stock openpilot?

DouglasSherk, 2023-11-21:
I can’t reproduce this on stock OP 0.9.5.

closed 2023-11-21 by DouglasSherk