commaai/openpilot #30511: Constant Cruise Faults '17 Chevy Volts

TheNathanielWatkins, 2023-11-22:
### Describe the bug

Since upgrading to any version 0.9.5 or greater, 2017 Chevy Volts get constant Cruise Faults, preventing any driver assist from working.  This happenes with stock `v0.9.5-release` as well as `v0.9.6.1` master, and also on Forks based off of v0.9.5+.

Discussion of the bug happened over the past couple of days in the GM channel of the official Discord, starting from this post: https://discord.com/channels/469524606043160576/524611823090008065/1175930009756639303

[image]
[image]
[image]
[image]

### Which car does this affect?

Chevrolet Volt 2017

### Provide a route where the issue occurs

c57d330d63109e79|2023-11-19--22-58-49--1

### openpilot version

0.9.5-release

### Additional info

To my knowledge, this issue is not happening to any other cars or even other years of Volt; at least it's not being reported in the channels I've been looking in to troubleshoot the issue.

TheNathanielWatkins, 2023-11-22:
Update, it seems that one of my images didn't upload properly.  Here it is again:
[image]

adeebshihadeh, 2023-11-24:
Can you upload the rlogs for a route with this issue? Can you also give this branch a try: `installer.comma.ai/commaai/fix-volt`?

adeebshihadeh, 2023-11-25:
Looks like this was caused by https://github.com/commaai/panda/pull/1542. I've reverted it, and it's going out to release as soon as the tests pass. cc @briskspirit

adeebshihadeh, 2023-11-28:
The fix has been pushed to release. Let us know if you have any more issues.

closed 2023-11-28 by adeebshihadeh

TheNathanielWatkins, 2023-11-30:
Thank you! The fix seems to have worked. I installed the current `master` branch and it was able to calibrate and drive for a few minutes without issue. 

Sorry I didn't give you more timely answers or testing; life's been busy. I'll chime in if any further issues pop up after I've been driven more.