commaai/openpilot #30514: Ford Explorer Unrecognized

adamgranted, 2023-11-23:
### Describe the bug

I have a 2020 Ford Explorer ST and it's showing as unrecognized in Openpilot. It looks like I'm not the only one, my Explorer came out of the dealership yesterday and according to @agatoff9, [they're seeing the same issue.](https://github.com/commaai/openpilot/discussions/30485).

Let me know how I can help or what info I can provide.

### Which car does this affect?

Ford Explorer ST 2020

### Provide a route where the issue occurs

NA

### openpilot version

0.9.5

agatoff9, 2023-11-23:
I had the same issue last week. I temp fixed the problem by loading a 6th gen explorer specific version of openpilot. I think it's intended to force the fingerprinting but I'm not sure except that it works.  Please let me know what I can do to help get this fixed.

adamgranted, 2023-11-23:
@agatoff9 I'll give it a shot. Can you send me the branch you used?

adamgranted, 2023-11-24:
I was able to fix this issue in forked version of v0.9.5. I found two missing ECU values and added them in the Ford values.py. This version is working fine for me now on my 2020 Ford Explorer ST: [https://github.com/adamgranted/openpilot/tree/fordexplorer_ecuFix](https://github.com/adamgranted/openpilot/tree/fordexplorer_ecuFix).

I just had to use installer.comma.ai/adamgranted/fordexplorer_ecuFix for the installer URL.

mkos11, 2023-11-25:
It seems like there might be an issue with the recognition of the 2020 Ford Explorer ST in the openpilot software. Here are some steps you can take to address and report this bug:

1. **Community Forums:**
   Visit the openpilot community forums or relevant channels where users discuss their experiences. Look for any existing threads related to the 2020 Ford Explorer ST. Other users might have shared insights or workarounds.

2. **Check for Updates:**
   Ensure that you are running the latest version of the openpilot software. Bugs are often addressed in newer releases, and updating might resolve the recognition issue.

3. **Collect Information:**
   Gather detailed information about the issue, including any error messages or unusual behavior you observe. Note down specific circumstances or patterns where the recognition problem occurs.

4. **Reach Out to openpilot Support:**
   If the issue persists, consider reaching out to openpilot's official support channels. Provide them with a clear description of the problem, the make and model of your car, and any relevant details. They might have specific steps or updates to address the recognition issue.

5. **Collaborate with the Community:**
   Collaborate with other users who are facing similar problems. If there's a consensus that this is a widespread issue, it can be more effectively communicated to the openpilot development team.

6. **Update Community Documentation:**
   If you find a solution or receive guidance from the openpilot community or support, consider updating relevant documentation or forum threads. This helps other users who might encounter the same problem in the future.

7. **Stay Informed:**
   Keep an eye on openpilot's official channels, such as GitHub repositories or release notes. The development team might release updates or fixes related to car model recognition.

Remember that these steps provide general guidance, and the specifics might vary based on the evolving nature of openpilot development. If there's a dedicated bug reporting system or process for openpilot, follow those guidelines to ensure that the development team receives the necessary information to investigate and address the issue.

adamgranted, 2023-11-25:
@mkos11 yup I found the issue and fixed it in my own fork; as notated in my last comment. 

I'm new to the community so whatever is the best method to report my findings let me know.

sshane, 2023-11-25:
Can you provide your dongle ID? It will show in the Device settings.

adamgranted, 2023-11-25:
@sshane yup, this is it right? b18d8435f1460528

sshane, 2023-11-28:
Would you like to make a PR @adamgranted?

David-prince1, 2023-11-30:
> I was able to fix this issue in forked version of v0.9.5. I found two missing ECU values and added them in the Ford values.py. This version is working fine for me now on my 2020 Ford Explorer ST: [...]

I tried this installer in my 2023 Explorer ST. Did not work. Any suggestions? Still showing car unrecognized.

David-prince1, 2023-12-01:
> I had the same issue last week. I temp fixed the problem by loading a 6th gen explorer specific version of openpilot. I think it's intended to force the fingerprinting but I'm not sure except that it works. Please let me know what I can do to help get this fixed.

How do I do that?

adamgranted, 2023-12-01:
> Would you like to make a PR @adamgranted?

I can. I'm not super familiar with Git or development in general but I'll try.

> I tried this installer in my 2023 Explorer ST. Did not work. Any suggestions? Still showing car unrecognized.
@David-prince1 Do you have a USB or Bluetooth OBD adapter? I just used Forscan (free trial) to identify the missing modules.

David-prince1, 2023-12-01:
@adamgranted Thanks for your response. I do not. Also not 100% sure what that is lol. Should I get an OBD adapter? What will that tell me.

sheaduncan, 2023-12-02:
All you had to do was go to useradmin.comma.ai

click on route
click on the first segment
in the drop-down in the left corner, select "CarParams" and there are all your FW versions.

adamgranted, 2023-12-03:
Thanks @sheaduncan, I'm new to the community and just went with what I knew.

sshane, 2023-12-07:
@David-prince1 what is your dongle id?

sshane, 2023-12-07:
Ah I see we got you merged in https://github.com/commaai/openpilot/pull/30637 @David-prince1

adamgranted, 2023-12-07:
>  @adamgranted any problems with openpilot?
@sshane My bad I forgot to do a PR. Looks like you got it in #30638?

Everything has been working fine for me. I'm still using my forked version.

There's only one issue that I've seen; it's happened twice. OpenPilot will say my Explorer is not recognized but that's only been after using the remote start. Once you turn the vehicle on and off again it works fine.

closed 2023-12-08 by sshane