commaai/openpilot #30516: Car Unrecognized

khokharashfaq, 2023-11-23:
### Describe the bug

Just purchased Comma 3 and installed the latest version of openpilot. I am getting car Unrecognized prompt soon after turning on the vehicle. 

### Which car does this affect?

Honda Civic EX 2019

### Provide a route where the issue occurs

No route available yet.

### openpilot version

0.9

sshane, 2023-11-25:
Can you provide your dongle ID? It will show in the Device settings.

khokharashfaq, 2023-11-26:
My dongle ID is 9ae7a9ee7fa08103.  Now that car unrecognized message is gone.  Do I need to have a SIM card for full self driving capability ? If yes, would t-mobile data service work ? ThanksSent from my iPhone

sshane, 2023-11-30:
Looks like you were missing your comma power.

No data is required for regular chill mode (not navigation-aware). However, if you want to use Navigate on openpilot, you need comma prime or prime lite where you provide your own SIM card: https://comma.ai/connect

closed 2023-11-30 by sshane